atomvm/AtomVM #1: bifs_hash.h not found

fermuch, 2018-06-16:
Hello! I've been trying to use AtomVM, but at compile time I get this error:

```
src/libAtomVM/bif.c:30:23: fatal error: bifs_hash.h: No such file or directory
compilation terminated.
```

I'm unsure where bifs_hash comes from.

bettio, 2018-06-18:
Did you install gperf on your system? gperf is required to generate bifs_hash.h.
On Ubuntu/Debian it can be installed with `apt-get install gperf`.
Let me know if you are using any other operating system if you need further help.

fermuch, 2018-06-18:
I do have gperf. I'm using archlinux, and I do have these packages:

```
extra/gperf 3.1-1
    Perfect hash function generator
extra/gperftools 2.7-1
    Fast, multi-threaded malloc and nifty performance analysis tools
```

bettio, 2018-06-18:
You need the first one to build AtomVM.
On my system:
```
$ gperf --version 
GNU gperf 3.1
```
There was also another problem that caused a similar issue with bifs_hash.h, I've also fixed it right now, so make also sure to update your clone.

Let me know if you have any other issue, in the worst case `VERBOSE=1 make` might be also helpful to debug further this kind of issues.

fermuch, 2018-06-18:
It now compiles! Thanks!

closed 2018-06-18 by fermuch