atomvm/AtomVM #21: SEGV when passing atoms through open_port (atom hash table corruption?)

fadushin, 2018-11-25:
I have some code that is passing a property list to a port, as follows:

`
open(Port, Params) ->
    open_port({spawn, "socket"}, [{proto, udp} | Params]).
`

In my driver initialization, I have the following code:

```
driver_status socket_driver_init(Context *ctx, term opts)
{
    driver_status ret;
    GlobalContext *glb = ctx->global;
    struct SocketDriverData *socket_driver_context = calloc(1, sizeof(struct SocketDriverData));

    ctx->native_handler = consume_socket_driver_mailbox;
    ctx->platform_data = socket_driver_context;
    
    if (!term_is_list(opts)) {
        fprintf(stderr, "opts is not a list\n");
        abort();
    }
    
    term proto = interop_proplist_get_value(opts, term_from_atom_string(glb, tag_proto_a));
    
    if (term_is_nil(proto)) {
        fprintf(stderr, "no proto in opts\n");
        abort();
    }
```

However, I am getting a segv when calling `term_from_atom_string(glb, tag_proto_a)` in the above snippet, where `tag_proto_a` is defined as:

`static const char *const tag_proto_a = "\x5" "proto";`

I can see that the `proto` atom is inserted into the Global Context's atom table in `parse_external_terms`:

https://github.com/bettio/AtomVM/blob/master/src/libAtomVM/externalterm.c#L69

However, by the time my port driver is initialized, the atom hash table seems to have corrupt data.  The crash is in de-referencing node->key, in `atomshashtable_get_value `:

```
unsigned long atomshashtable_get_value(const struct AtomsHashTable *hash_table, const AtomString string, unsigned long default_value)
{
    unsigned long hash = sdbm_hash(string, atom_string_len(string));
    long index = hash % hash_table->capacity;

    const struct HNode *node = hash_table->buckets[index];
    while (node) {
        if (atom_are_equals(string, node->key)) {
            return node->value;
        }

        node = node->next;
    }

    return default_value;
}
```

Here is what node looks like:

```
node	const HNode *	0x2c	0x000000000000002c
next	HNode *	NULL	
next	HNode *	NULL	
key	AtomString	0x0	
value	unsigned long	
key	AtomString	0x0	
value	unsigned long	
```

Here is the stack trace:

```
Thread 1 Queue : com.apple.main-thread (serial)
#0	0x000000010000245c in atomshashtable_get_value at /work/src/github/fadushin/AtomVM/src/libAtomVM/atomshashtable.c:115
#1	0x000000010001a800 in globalcontext_insert_atom at /work/src/github/fadushin/AtomVM/src/libAtomVM/globalcontext.c:149
#2	0x000000010002fd3d in term_from_atom_string at /work/src/github/fadushin/AtomVM/src/libAtomVM/context.h:188
#3	0x000000010002f78e in socket_driver_init at /work/src/github/fadushin/AtomVM/src/platforms/generic_unix/socket_driver.c:77
#4	0x00000001000308d5 in platform_open_port at /work/src/github/fadushin/AtomVM/src/platforms/generic_unix/sys.c:209
#5	0x000000010002c3ac in nif_erlang_open_port_2 at /work/src/github/fadushin/AtomVM/src/libAtomVM/nifs.c:285
#6	0x0000000100014a11 in context_execute_loop at /work/src/github/fadushin/AtomVM/src/libAtomVM/opcodesswitch.h:2137
#7	0x0000000100001fbe in main at /work/src/github/fadushin/AtomVM/src/main.c:84
#8	0x00007fff7c3dd08d in start ()
#9	0x00007fff7c3dd08d in start ()
```

Is it possible that the memory referenced during the initial population of the atom hash table (from external terms) is no longer valid?  Should the hash table keep a _copy_ of these atom strings?

Thanks!

fadushin, 2018-11-25:
See https://github.com/fadushin/AtomVM/commit/6d6957997184e8a8d3da9e07d78962708e168792 for an example, using the test module:

```
-module(smoke).

-export([start/0]).

start() ->
    State = gen_udp:open(0),
    console:puts("ok\n").
```

bettio, 2018-11-27:
Thanks, I'll investigate it

bettio, 2018-11-27:
The problem is how the port driver memory is managed, so right now the garbage collector might run while a message is being processed and terms that are on the native C stack might be invalid after the garbage collection.

To avoid this kind of issues memory_ensure_free(ctx) is usually called before allocating new items to the stack, however this approach doesn't scale well and the resulting C code might be boring.
A different approach to this issue should be used, anyway right now the easiest work around is to avoid it by changing default heap size.

This might workaround your problem:
```
diff --git a/src/libAtomVM/context.c b/src/libAtomVM/context.c
index 8861043..830a881 100644
--- a/src/libAtomVM/context.c
+++ b/src/libAtomVM/context.c
@@ -26,7 +26,7 @@
 #include "opcodesswitch.h"
 #undef IMPL_EXECUTE_LOOP
 
-#define DEFAULT_STACK_SIZE 8
+#define DEFAULT_STACK_SIZE 65536
 
 Context *context_new(GlobalContext *glb)
 {
```

closed 2018-12-14 by bettio