atomvm/AtomVM #24: Minimum hardware requirements (Arduino support?)

Serkan-devel, 2018-12-02:
Considering that this VM should be small, on what hardware platform is it still possible to run on?
This project looks a little similar to [MicroPython](https://micropython.org/) from it's possible use case.

* How much of a performance improvement does AtomVM have over BEAM?
* Is it possible to run it on microcontrollers, like the Arduino Uno?

bettio, 2018-12-03:
1. AtomVM does not have execution speed improvements over BEAM, to be honest AtomVM runs slower. The aim is to run with a lower memory and flash footprint (with a lower speed).
2. Yes, with some effort, but it wouldn't be usable for real world purposes, Arduino Uno has a 8 bits CPU with less than 10 KB of RAM and a harvard architecture. It will be possible to run on microcontrollers with a reasonable amount of memory, such several ARM Cortex-M.
3. Some will be optional, such as floating point values, unlimited precision integers, clustering features, code reloading others will not be implemented, such as HiPE, advanced debugging and profiling features.

jeremytregunna, 2019-01-12:
@bettio so approximately how much memory and flash would you need to port atomvm to another board? I was thinking about doing that with one of mine this weekend until I saw your post saying 10kb was not reasonable (mine has 8kb ram, 24kb flash, but is a Cortex-M0+).

ponyatov, 2020-09-29:
> I was thinking about doing that with .. 8kb ram, 24kb flash, but is a Cortex-M0+.
You can try to follow some other way, via C code generation in Elixir -- it does not need any side runtimes and can be truly HardRT. I can't say it is simple, but in some cases, it can be a variant to use friendly language to write code in other platform/employer-forced language stacks.

UncleGrumpy, 2023-11-12:
For anyone following this issue, after some extensive work on the STM32 platform, I can say the minimum flash requirement would be 512K of flash. For devices with this minimum amount of flash the compiler optimization must be `-Os` to optimize for size. On STM32 devices this will leave 128K of flash available for `.avm` packed beam applications. The absolute minimum RAM requirement has not been established yet, but as far as STM32 devices go, any with 512K or more of flash also have adequate RAM to be functional. 128K of RAM is enough to run the [atomvm_benchmark](https://github.com/pguyot/atomvm_benchmark) tests, including sodoku puzzle generation and solutions.