atomvm/AtomVM #30: Web assembly

olafura, 2018-12-07:
[image]

Here is what I did:
Apply this patch with `patch -p1 < patch`
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index b339d48..83d21fc 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -4,7 +4,7 @@ project (AtomVM)
 
 add_subdirectory(src)
 add_subdirectory(tests)
-add_subdirectory(tools/packbeam)
+# add_subdirectory(tools/packbeam)
 
 if (CMAKE_BUILD_TYPE STREQUAL "Coverage")
     set(CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/CMakeModules)
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index b8fc83f..f152599 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -32,13 +32,15 @@ set(
 if(CMAKE_COMPILER_IS_GNUCC)
     set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -Wall -pedantic -Wextra -ggdb")
 endif()
+# set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -s FORCE_FILESYSTEM=1 -s ENVIRONMENT=node --preload-file hello_world.beam")
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -v --embed-file hello_world.beam")
 
 include_directories(${CMAKE_CURRENT_BINARY_DIR} libAtomVM/)
 
-add_executable(AtomVM main.c)
-target_link_libraries(AtomVM libAtomVM libAtomVM${PLATFORM_LIB_SUFFIX})
-set_property(TARGET AtomVM PROPERTY C_STANDARD 99)
+add_executable(AtomVM.html main.c)
+target_link_libraries(AtomVM.html libAtomVM libAtomVM${PLATFORM_LIB_SUFFIX})
+set_property(TARGET AtomVM.html PROPERTY C_STANDARD 99)
 
 if (CMAKE_BUILD_TYPE STREQUAL "Coverage")
-    set_target_properties(AtomVM PROPERTIES COMPILE_FLAGS "-O0 -fprofile-arcs -ftest-coverage")
+    set_target_properties(AtomVM.html PROPERTIES COMPILE_FLAGS "-O0 -fprofile-arcs -ftest-coverage")
 endif()
diff --git a/src/libAtomVM/CMakeLists.txt b/src/libAtomVM/CMakeLists.txt
index 3fc09f7..252efe1 100644
--- a/src/libAtomVM/CMakeLists.txt
+++ b/src/libAtomVM/CMakeLists.txt
@@ -84,3 +84,4 @@ set_property(TARGET libAtomVM PROPERTY C_STANDARD 99)
 if (CMAKE_BUILD_TYPE STREQUAL "Coverage")
     set_target_properties(libAtomVM PROPERTIES COMPILE_FLAGS "-O0 -fprofile-arcs -ftest-coverage")
 endif()
+set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -I${HOME}/dev/zlib")
```

Run this to get the make files
`emconfigure cmake -D CMAKE_C_COMPILER=emcc -D CMAKE_CXX_COMPILER=emcc -D ZLIB_INCLUDE_DIR=$HOME/dev/zlib -D ZLIB_LIBRARY=$HOME/dev/zlib/libz.a .`

Then run:
`emmake make`

Then add `arguments: ['/hello_world.beam'],` to Atom.html after `var Module = {`

This is such a cool project ;)

bettio, 2018-12-07:
This is totally unexpected and cool :)

So please send a PR with a new platform for webassembly, the webassembly platform should contain all the required code and builsystem files needed to run on a browser.
It might be interesting to support loading .avm/.beam using a file selector or from a remote http, I think it might be also a good solution for testing purposes :)

If you have any question there is also a slack channel (#atomvm) on elixir lang slack.

bettio, 2023-07-28:
[emscripten](https://github.com/atomvm/AtomVM/tree/master/src/platforms/emscripten) platform has been added.

closed 2023-07-28 by bettio