atomvm/AtomVM #41: Sending a nested tuple crashes AtomVM (abort)

fadushin, 2018-12-19:
The following code will crash AtomVM:

```
-module(pingpong).

-export([start/0, ping/2, pong/1]).

start() ->
    Console = console:start(),
    Pong = spawn(pingpong, pong, [Console]),
    spawn(pingpong, ping, [Console, Pong]),
    receive
        ok -> ok
    end.

ping(Console, Pong) ->
    Pong ! {self(), {ping, foo}},      %% <-- crash here when putting inner tuple
    receive
        {Pong, pong} -> 
            console:puts("+"),
            ping(Console, Pong)
    end.

pong(Console) ->
    receive
        {Ping, {ping, _Foo}} -> 
            console:puts("-"),
            Ping ! {self(), pong},
            pong(Console)
    end.
```

With the stack trace:

```
#2	0x00007fff5de241c9 in abort ()
#3	0x00000001000194ae in term_put_tuple_element at /work/src/github/fadushin/AtomVM/src/libAtomVM/term.h:619
#4	0x0000000100013d48 in context_execute_loop at /work/src/github/fadushin/AtomVM/src/libAtomVM/opcodesswitch.h:2054
#5	0x00000001000017ee in main at /work/src/github/fadushin/AtomVM/src/main.c:84
#6	0x00007fff5dd7c08d in start ()
#7	0x00007fff5dd7c08d in start ()
```

Here are the compiled BEAM instructions for this code:

```
{module, pingpong}.  %% version = 0

{exports, [{module_info,0},{module_info,1},{ping,2},{pong,1},{start,0}]}.

{attributes, []}.

{labels, 20}.

{function, start, 0, 2}.
  {label,1}.
    {line,[{location,"../pingpong.erl",5}]}.
    {func_info,{atom,pingpong},{atom,start},0}.
  {label,2}.
    {allocate_zero,1,0}.
    {line,[{location,"../pingpong.erl",6}]}.
    {call_ext,0,{extfunc,console,start,0}}.
    {test_heap,2,1}.
    {move,{x,0},{y,0}}.
    {put_list,{y,0},nil,{x,2}}.
    {move,{atom,pong},{x,1}}.
    {move,{atom,pingpong},{x,0}}.
    {line,[{location,"../pingpong.erl",7}]}.
    {call_ext,3,{extfunc,erlang,spawn,3}}.
    {test_heap,4,1}.
[... 395 lines cut ...]
move/2 b, x0
line/1: 4
call_ext/2, arity=3, index=1
line/1: 5
label/1 label=3
loop_rec/2, dreg=0
label/1 label=5
wait/1
label/1 label=12
allocate_zero/2 stack_need=2, live=1
move/2 23, y1
line/1: 11
label/1 label=13
loop_rec/2, dreg=0
label/1 label=15
wait/1
label/1 label=7
allocate_heap_zero/3 stack_need=2, heap_need=3, live=2
bif0/2 bif=2, dreg=2
move/2 33, y1
move/2 23, y0
put_tuple/2 size=2, dest=x1
put/2 elem=0, value=0x43
put/2 elem=1, value=0x1005070aa
```

fadushin, 2018-12-24:
I believe this is due to an issue with garbage collection -- the original tuple is allocated here:

https://github.com/bettio/AtomVM/blob/ecc81928b6b039e873d838b81e8cd7eb65b57417/src/libAtomVM/opcodesswitch.h#L2050

but the subsequent call to the DECODE_COMPACT_TERM macro on line 2061 may call `module_load_literal`, which in turn may call `memory_ensure_free`, which may run the GC.

Here is a branch with a workaround:

https://github.com/fadushin/AtomVM/tree/send-tuple-workaround

I realize the memory subsystem is under re-design, so I will not submit the branch for a PR, but am only providing it for information purposes, or in case anyone is in need of a temporary workaround.

BTW, I found I had to remove the if clause:

```
    if (context_memory_size(c) > size) {
        if (UNLIKELY(memory_gc(c, context_memory_size(c)) != MEMORY_GC_OK)) {
            //TODO: handle this more gracefully
            fprintf(stderr, "Unable to allocate memory for GC\n");
            abort();
        }
    }
```

in `memory_ensure_free`, as `context_memory_size(c)` returns the allocated size of the heap (and stack) and will almost always be > size.  So I am a little confused by this logic?  The tests pass without this if clause, but I don't know if I am missing something obvious.

bettio, 2019-01-02:
The new garbage collector fixes this issue, so I'm closing this issue. Thanks for your bug report :)

closed 2019-01-02 by bettio