atomvm/AtomVM #43: Consider Gimli as general hash function

mko-io, 2018-12-21:
Hi just found this cool project, good work. One of the dependency is GNU Perfect Hash Function, I'm not sure the use for it, whether its the general hash function used internally for phash/1 or phash2/1, or for other purpose.
I have discovered a [gimli](https://gimli.cr.yp.to) permutation function used in [libhydrogen](https://www.libhydrogen.org)
A good selling point for gimli is that it perform well across all hardware spec, in their [reference implementation](https://gimli.cr.yp.to/impl.html) cortex-M3 and cortex-M4 is included(only depend on libopencm3)
Make use of it could shrink the dependency even further

bettio, 2018-12-21:
gperf is a build time dependency: it takes a list of strings and it produces as output a C source file. The nifs and bifs hash table is stored in a read only memory area so it doesn't take RAM.

mko-io, 2018-12-21:
Thanks for the explanation, I'm closing this now because it's not related to application hashing function

closed 2018-12-21 by mko-io