atomvm/AtomVM #55: compilation for esp32

ssachse, 2019-01-22:
Hello, I am trying to compile for the esp32. Tried both version of IDF stable and current. Both don't compile the recent version. after running make, I always get:

<code>
GENCONFIG
*
* Restart config...
*
*
* Security features
*
Require signed app images (SECURE_SIGNED_APPS_NO_SECURE_BOOT) [N/y/?] (NEW) aborted!

Console input/output is redirected. Run 'make oldconfig' to update configuration.

make: *** No rule to make target `AtomVM/src/platforms/esp32/build/include/config/auto.conf', needed by `AtomVM/src/platforms/esp32/build/bootloader/bootloader.bin'.  Stop.
</code>

fadushin, 2019-01-22:
Can you clarify:

* What OS are you running?
* What version of the ESP-IDF are you using?

At this point, AtomVM has been using ESP-IDF 3.1.1, but I have run some code on 3.1.2, as well.

ESP-IDF releases can be found here:

https://github.com/espressif/esp-idf/releases

ssachse, 2019-01-28:
I am using macOS Mojave and tried with ESP-IDF 3.1.2

fadushin, 2019-01-28:
Same here.  Can you try the instructions here:

http://blog.dushin.net/2018/11/running-atomvm-on-macos-and-an-esp32/

bettio, 2019-02-21:
@ssachse Have you been able to build it?

ssachse, 2019-02-25:
@bettio sorry for not answering earlier, but I got a real bad flu here in Germany .... totally offline ;-( Now all compiles really well, thanx

closed 2019-02-25 by ssachse