atomvm/AtomVM #88: esp32 - receive with timeout > 9 does never return

spellauge, 2019-04-12:
the pgm

-module (r).
-export([start/0]).

start() ->
    MSecs = 10,
    console:puts("good night ..\n"),
    receive
        after MSecs -> ok
    end,
    console:puts("good morning\n")
    .

will output
Starting: r.beam...
---
good night ..
E (10305) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
...
which means, the task hangs in an infinite loop without any event,mbx,.. related activity. this seems to bre true for any MSecs > 9.

the same pgm works fine for MSecs < 10. i.e. it will output
Starting: r.beam...
---
good night ..
good morning

the same pgm works fine in AtomVM running under linux for any MSecs > 0

fadushin, 2020-10-02:
Changing code to use console:print (instead of puts), the code seems to work now.

```
.../AtomVM/build> ./src/AtomVM r.beam 
good night ..
good morning
Return value: ok
```

There have been several changes to the scheduler and sys.c module (referenced above) which may have addressed this bug.  Recommend closing.

elsbiet, 2021-04-10:
code runs fine now - even with console.puts

closed 2021-04-10 by elsbiet