atomvm/AtomVM #92: memory leak (related to string/list handling)

elsbiet, 2019-04-16:
there seems to be a memory leak (possibly related to string/list handling).

my test case:

-module (ht).
-export([start/0]).

start() ->
    console:puts("start\n"),
    loop(0),
    console:puts("done\n"),
    ok
.

loop(581) ->
    console:puts("loop done!\n");
loop(N) ->
    S  = "N=" ++ integer_to_list(N) ++ "\n",
    S = t(S),
    loop(N+1)
.

t(S) -> S
.

compile, pack and run under linux as

ulimit -Sv 5127
~/esp-erl/AtomVM/build/src/AtomVM ht.avm

will result in (at least on my machine):
start
loop done!
Aborted                 ~/esp-erl/AtomVM/build/src/AtomVM ht.avm

replacing 581 by 582 will produce the same result while replacing it by 583 will result in
start
Aborted                 ~/esp-erl/AtomVM/build/src/AtomVM ${F}.avm

replacing 581 by 580 

will result in:
start
loop done!
done
Return value: ok

the behaviour is even worse (lower threshold) when the pgm runs on my esp32.

i would try to find and fix the root cause(s?) of this problem but i'm afraid, i'm not sufficient familiar with the design principles of atomvm.

bettio, 2019-04-20:
Thanks for reporting this issue, I will investigate what is happening here.
I'll let you know in next few days :)

bettio, 2019-04-21:
I can't reproduce it right now. are you able to reproduce this issue on a clean build from the latest master version?

elsbiet, 2019-04-21:
after cloning the current master i could reproduce the issue.

- did you limit the memory (-Sv 5127) before starting AtomVM?
- maybe on your system it might be necessary to increase the loop count (e.g. by 10000000). 

erl-pgm:

-module (ht).
-export([start/0]).

start() ->
    console:puts("start\n"),
    loop(0),
    console:puts("done\n"),
    ok
    .

loop(585) ->
    console:puts("loop done!\n");
loop(N) ->
    S = "N=" ++ integer_to_list(N) ++ "\n",
    S = t(S),
    loop(N+1)
    .

t(S) -> S
    .

compile and pack

under linux

ulimit -Sv 5127                               # limit memory
$AtomVMHOME/build/src/AtomVM ht.avm

produces 

start
.. 28641 Aborted                 (core dumped) $AtomVMHOME/build/src/AtomVM ht.avm

inspecting the core-file (gdb $AtomVMHOME/build/src/AtomVM core)
..
Program terminated with signal SIGABRT, Aborted.
#0  0x00007f62fd37f8bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) bt
#0  0x00007f62fd37f8bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007f62fd36a535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00005571f150789a in context_execute_loop (ctx=0x5571f20a0f10, mod=0x5571f20a09f0, function_name=0x5571f1531070 "start", arity=0)
    at /home/splge/esp-erl-1019-04-21/AtomVM/src/libAtomVM/opcodesswitch.h:1171
#3  0x00005571f1502679 in main (argc=2, argv=0x7ffca296ae18) at /home/splge/esp-erl-1019-04-21/AtomVM/src/main.c:86

bettio, 2019-05-12:
I think #95 has fixed this issue too :)
Thanks for reporting :)

closed 2019-05-12 by bettio

elsbiet, 2019-05-12:
thanks for fixing.

i shall have a look at it when i find enough time to do it in a reasonable way - as i told you before, currently i'm involved in another, totally unrelated hobby project which consumes nearly all of (if not more than) my leisure time until end of june.
be sure i shall be back.