atomvm/AtomVM #96: memory leak?

elsbiet, 2019-04-19:
the program

-module (ri).
-export([start/0 ]).

repeat(_,0) -> ok;
repeat(F,N) -> F(N), repeat(F, N - 1).

start() -> repeat(
             fun(N) ->
                     ok=test({N})
             end,
             10000000),
           ok
           .

test(_S) -> ok.

when run under linux as
ulimit -Sv 5127      
~/esp-erl/AtomVM/build/src/AtomVM ri.avm

results in
7324 Aborted                 ~/esp-erl/AtomVM/build/src/AtomVM ri.avm

examinig the stack trace using gdb yields

 gdb ~/esp-erl/AtomVM/build/src/AtomVM core
..
(gdb) bt
#0  0x00007f8ce8b8d8bb in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007f8ce8b78535 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x0000556b1aae487a in context_execute_loop (ctx=0x556b1adfff70, mod=0x556b1ae00b70, function_name=0x556b1ab0d070 "start", arity=0)
    at /home/splge/esp-erl/AtomVM/src/libAtomVM/opcodesswitch.h:1171
#3  0x0000556b1aadf659 in main (argc=2, argv=0x7fffda875728) at /home/splge/esp-erl/AtomVM/src/main.c:86

replacing ok=test({N}) by ok=test([N]) produces exactly the same result.

replacing 10000000 by 1000  or replacing test[{N}) by test(N) results in Return value: ok

bettio, 2019-04-20:
Thanks for reporting, I will try to reproduce this issue.

bettio, 2019-05-12:
#95 should have fixed this issue.
Let me know if you still experience any issue.

closed 2019-05-12 by bettio