atomvm/AtomVM #98: sending message to a registered process via it's name does not work.

elsbiet, 2019-05-16:
running the pgm

-module(pr).
-export([start/0, ping/1, pong/0]).
ping(0) ->
   pong ! finished,
    erlang:display("ping finished");

ping(N) ->
    erlang:display("ping started"),
    pong ! {ping, self()},
    erlang:display("ping sent"),
    receive
        pong ->
            erlang:display("Ping received pong")
    end,
    ping(N - 1).

pong() ->
    erlang:display("pong started"),
    receive
        finished ->
            erlang:display("Pong finished");
        {ping, Ping_PID} ->
            erlang:display("Pong received ping"),
            Ping_PID ! pong,
            pong()
    end.

start() ->
    register(pong, spawn(pr, pong, [])),
    spawn(pr, ping, [3]).

results in:
"pong started"
"ping started"
term is not a pid: a4b
"ping sent"
Hang detected

after replacing 

pong ! Msg,

 by 

P = whereis(pong), P ! Msg,

the expected output 
"pong started"
"ping started"
"ping sent"
"Pong received ping"
"pong started"
"Ping received pong"
"ping started"
"ping sent"
"Pong received ping"
"pong started"
"Ping received pong"
"ping started"
"ping sent"
"Pong received ping"
"pong started"
"Ping received pong"
"ping finished"
"Pong finished"
Return value: <0.3.0>

is displayed.

bettio, 2019-09-04:
Update: this bug is still here (tested on 1939abfaad3faf64f982eee39d5db5bd146f279e)

elsbiet, 2021-04-28:
still unresolved

fadushin, 2024-02-18:
This issue is still open, if you consider the send operator (`!`), which the original poster used.  #631 addresses the Erlang:send nif, not the `OP_SEND` opcode.

Here is a simpler example to reproduce the error:

```
-module(test).

-export([start/0]).

start() ->
    Pid = spawn_opt(fun loop/0, []),
    register(foo, Pid),
    foo ! {ping, self()},
    receive
        pong ->
            ok
    after 1000 ->
        timeout
    end.

loop() ->
    receive
        {ping, Pid} ->
            Pid ! pong,
            loop()
    end.
```

On OTP:

```
Eshell V12.3.2.15  (abort with ^G)
1> test:start().
ok
```

On AtomVM:

```
> atomvm test.beam
Unsupported line_ref tag: 0
Return value: timeout
```

I believe the issue may relate to the current implementation of the `send` opcode:

```
            case OP_SEND: {
                #ifdef IMPL_CODE_LOADER
                    TRACE("send/0\n");
                #endif

                #ifdef IMPL_EXECUTE_LOOP
                    int local_process_id = term_to_local_process_id(x_regs[0]);
                    TRACE("send/0 target_pid=%i\n", local_process_id);
                    TRACE_SEND(ctx, x_regs[0], x_regs[1]);
                    globalcontext_send_message(ctx->global, local_process_id, x_regs[1]);

                    x_regs[0] = x_regs[1];
                #endif
                break;
            }
```

As you can see, we are assuming the term in the 0th x-register is a Pid, when in fact it could be an atom.

UncleGrumpy, 2024-02-18:
The oversight of missing OP_SEND opcode is finally addressed in #1047.

PR #631 (2023-06-28): Allow sending message to registered process name
Changes `erlang:send/2` to accept a pid or registered process name. Updates `tests/erlang_tests/test_send_nif_and_echo.erl` to also test sending directly to the registered name, and `tests/erlang_tests/test_send.erl` to test sending to an an unregistered `atom` name.

closes #98

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit e7ac8fc2 (2024-02-22): Merge pull request #1047 from UncleGrumpy/send_registered

Allow sending messages to registered process with `!`

Enables sending messages to registered processes using the shorthand `!`
operator. If a message is sent to an atom name that is not registered using the
`!` operator a `badarg` run-time error occurs, matching OTP behaviour.

Closes #1011
Closes #98

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit eab4e320 (2024-02-22): Allow sending messages to registered process with `!`

Enables sending messages to registered processes using the shorhand `!`
operator. If a message is sent to an atom name that is not registered using the
`!` operator a `badarg` run-time error occurs, matching OTP behaviour.

closes #1011
closes #98
PR #1047 (2024-02-22): Allow sending messages to registered process with `!`
Enables sending messages to registered processes using the shorthand `!` operator. If a message is sent to an atom name that is not registered using the `!` operator a `badarg` run-time error occurs, matching OTP behaviour.

Closes #1011
Closes #98

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2024-02-28 by bettio