atomvm/AtomVM #99: implicit declaration error

tomash, 2019-05-22:
Ubuntu 18.04
esp-idf version 3.1.1, 3.1.4 (both confirmed to work with my ESP32-WROOM-32D, the example hello_world program flashes and outputs as expected to `make monitor`, both also working with mruby/c)

Trying to follow blogpost
https://medium.com/@Bettio/atomvm-how-to-run-elixir-code-on-a-3-microcontroller-b414773498a6
with a little help from
http://blog.dushin.net/2018/11/running-atomvm-on-macos-and-an-esp32/
(as the Medium blogpost does not mention building AtomVM itself)

The `make flash` in `AtomVM/src/platforms/esp32` fails with:

```
/home/tomek/esp/elixir-esp32/AtomVM/src/platforms/esp32/main/spidriver.c: In function 'spidriver_transfer_at':
/home/tomek/esp/elixir-esp32/AtomVM/src/platforms/esp32/main/spidriver.c:123:15: error: implicit declaration of function 'spi_device_polling_transmit' [-Werror=implicit-function-declaration]
     int ret = spi_device_polling_transmit(spi_data->handle, &spi_data->transaction);
               ^
/home/tomek/esp/elixir-esp32/AtomVM/src/platforms/esp32/main/spidriver.c:123:9: warning: unused variable 'ret' [-Wunused-variable]
     int ret = spi_device_polling_transmit(spi_data->handle, &spi_data->transaction);
         ^
cc1: some warnings being treated as errors
/home/tomek/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'spidriver.o' failed
make[1]: *** [spidriver.o] Error 1
/home/tomek/esp/esp-idf/make/project.mk:468: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
tomek@tomek-ux32ln:~/esp/elixir-esp32/AtomVM/src/platforms/esp32$ 
```

I put the entire build log from `make flash` on pastebin:
https://pastebin.com/kwesaMjs

I'm not finding `spi_device_polling_transmit` mention in esp-idf 3.1(.1) docs, it's mentioned in docs for 3.2 onwards, so maybe there is a specific 3.2.x esp-idf version that AtomVM builds with?

What's funny is I got past this one with some weird state of esp-idf master git-checkout'd to v3.1.4, but then it failed later on trying to build a component that was a leftover folder from master (3.3?), not present in 3.1 release.

tomash, 2019-05-22:
All right, got it to build and flash with esp-idf 3.2, cloned with `git clone -b v3.2 --recursive https://github.com/espressif/esp-idf.git`. No other tag seems to work: master fails due to missing `apps/sntp/sntp.h` (it's been moved), 3.1 fail for reason shown above.

So this issue is now a suggestion to explicitly state in readme that esp-idf version 3.2 is the only supported one by atomvm master at the moment.

tomash, 2019-05-22:
This is another issue but basically after building and flashing the onboard led is not blinking every 1 second and `make monitor` shows that the board is stuck in boot loop due to missing `timer` module:

```
Found AVM partition: size: 1048576, address: 0x110000
Booting file mapped at: 0x3f420000, size: 1048576
Starting: Elixir.Blink.beam...
---
Failed to open module: timer.beam
Warning: module timer cannot be resolved.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4011ecb6  PS      : 0x00060f30  A0      : 0x800d2d1d  A1      : 0x3ffbb270  
0x4011ecb6: context_execute_loop at /home/tomek/esp/elixir-esp32/AtomVM/src/libAtomVM/opcodesswitch.h:816

A2      : 0x3ffbf530  A3      : 0x00000068  A4      : 0x00000002  A5      : 0x00000000  
A6      : 0x3f420128  A7      : 0x00000001  A8      : 0x8011ecb4  A9      : 0x3ffbb140  
A10     : 0x00000000  A11     : 0x3ffbf474  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3f4200d2  A15     : 0x3f4200d8  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x4011ecb6:0x3ffbb270 0x400d2d1a:0x3ffbb2c0 0x400d110a:0x3ffbb2f0 0x4008e841:0x3ffbb310
0x4011ecb6: context_execute_loop at /home/tomek/esp/elixir-esp32/AtomVM/src/libAtomVM/opcodesswitch.h:816

0x400d2d1a: app_main at /home/tomek/esp/elixir-esp32/AtomVM/src/platforms/esp32/main/main.c:73

0x400d110a: main_task at /home/tomek/esp/esp-idf/components/esp32/cpu_start.c:494

0x4008e841: vPortTaskWrapper at /home/tomek/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6144
ho 0 tail 12 room 4
load:0x40078000,len:10180
load:0x40080400,len:6640
entry 0x40080760
I (31) boot: ESP-IDF v3.2 2nd stage bootloader
I (31) boot: compile time 11:55:01
I (31) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot:  3 main.avm         RF data          01 01 00110000 00100000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1bc24 (113700) map
I (141) esp_image: segment 1: paddr=0x0002bc4c vaddr=0x3ffb0000 size=0x0352c ( 13612) load
I (147) esp_image: segment 2: paddr=0x0002f180 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/tomek/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (148) esp_image: segment 3: paddr=0x0002f588 vaddr=0x40080400 size=0x00a88 (  2696) load
I (158) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x7f394 (521108) map
0x400d0018: _stext at ??:?

I (348) esp_image: segment 5: paddr=0x000af3b4 vaddr=0x40080e88 size=0x13084 ( 77956) load
I (392) boot: Loaded app from partition at offset 0x10000
I (393) boot: Disabling RNG early entropy source...
I (393) cpu_start: Pro cpu up.
I (397) cpu_start: Starting app cpu, entry point is 0x400811a4
0x400811a4: call_start_cpu1 at /home/tomek/esp/esp-idf/components/esp32/cpu_start.c:246

I (388) cpu_start: App cpu up.
I (407) heap_init: Initializing. RAM available for dynamic allocation:
I (414) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (420) heap_init: At 3FFB94F8 len 00026B08 (154 KiB): DRAM
I (426) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (433) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (439) heap_init: At 40093F0C len 0000C0F4 (48 KiB): IRAM
I (445) cpu_start: Pro cpu start user code
I (128) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```

tomash, 2019-05-22:
same with provided example file, i.e.:
```$IDF_PATH/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect  0x110000 examples/erlang/esp32/blink.avm```

tomash, 2019-05-22:
hello world example runs fine. So it seems that the issue is limited to timer module.

```
Found AVM partition: size: 1048576, address: 0x110000
Booting file mapped at: 0x3f420000, size: 1048576
Starting: hello_world.beam...
---
Hello World
AtomVM finished with return value = ok
going to sleep forever..
```

elsbiet, 2019-05-24:
try to replace timer:sleep by avm_timer:sleep and add avm_timer to your .avm-file

bettio, 2022-03-28:
Old issue, no longer an issue.

closed 2022-03-28 by bettio