atomvm/AtomVM #110: esp32 - blinky.avm not running

flbauer, 2019-09-06:
I am not getting any payload program running.

Followed: 
http://blog.dushin.net/2018/11/running-atomvm-on-macos-and-an-esp32/

What is working:
- AtomVM build works
- Flashing of bootloader works
- Compiling and packbin of blink.erl yields blink.avm 
- Flashing of a compiled blink.avm with the esptool.py seems to work but...
- program is never started

In Serial monitor I see:
`Starting avm_calendar.beam...
...
No start/0 function found.
AtomVM finished with return value = []
going to sleep forever`

I am sure I have a AtomVM on the board but for some reason the target program is never executed.
Hints?

bettio, 2019-09-07:
I think the problem is about how .beam files are packed together, AtomVM looks for start/0 function (the entry point) in the module that is marked as the startup module.
Right now PackBEAM is setting the first given module as the startup module, which in you case was by mistake avm_calendar.beam.
On your side if you want to get blink.avm working you should make sure that blink.beam is the first module packed into the .avm file.
On our side I think we should add a switch to PackBEAM so we can explicitly set the startup module.

bettio, 2019-09-07:
See also Issue #111

flbauer, 2019-09-07:
**Thank you for your explanation!** It helped and I got blink.avm to work now.
However I needed to do some  hackery:

- needed to switch to GPIO 4 instead of 2
- I had to create symbolic links of the .hrl files in the directories where the sources reside

Actually I would like to understand  the directory structure and the build tooling better. 

Although there are lots of CMakeFiles I get only errors invoking cmake
`CMake Error at CMakeLists.txt:7 (include):
  include could not find load file:

    BuildErlang

CMake Error at CMakeLists.txt:9 (pack_runnable):
  Unknown CMake command "pack_runnable".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.10)
`

closed 2019-10-24 by flbauer