atomvm/AtomVM #112: Implement packBEAM in erlang

bettio, 2019-09-07:
packBEAM should be also implemented in Erlang, so it can be integrated into existing Erlang and Elixir tooling in a easier way.

pfspontus, 2019-10-17:
@bettio when you say

> implemented in Erlang

Do you specifically mean Erlang or would an Elixir implementation do?

bettio, 2019-10-19:
> Do you specifically mean Erlang or would an Elixir implementation do?

I think Erlang would be better here because we don't require Elixir stuff and we are just keeping it optional. Also it is quite easy to use Erlang code from Elixir, so writing Elixir tooling using Erlang code  would be fairly simple.

fadushin, 2020-05-22:
See https://github.com/fadushin/atomvm_packbeam and https://github.com/fadushin/atomvm_rebar3_plugin

bettio, 2022-03-28:
This feature has been implemented.

closed 2022-03-28 by bettio