atomvm/AtomVM #113: esp32 - get_tcp

flbauer, 2019-09-08:
Is it already possible to create a tcp server in AtomVM?
What would be necessary to implement it?

fadushin, 2019-09-09:
It is on its way.  There is an implementation at https://github.com/fadushin/AtomVM/tree/gen-tcp, but it needs to be retrofitted to some redesign in the socket driver.

flbauer, 2019-09-10:
Trying to make tcp_client run.

The following attempt did not work: 

```
start() ->
    erlang:display(start),
    try
        Self = self(),
        Config = [
            {sta, [
                {ssid, "XXXXXXXXXXXXXX"},
                {psk,  "YYYYYYYYYYYYYY"},
                {connected, fun() -> Self ! connected end},
                {got_ip, fun(IpInfo) -> Self ! {ok, IpInfo} end},
                {disconnected, fun() -> Self ! disconnected end}
            ]}
        ],
        case network_fsm:start(Config) of
            ok ->
                wait_for_network();
            Error ->
                erlang:display(Error)
        end
    catch
        _:E ->
            erlang:display(E)
    end.

wait_for_network() ->
    erlang:display(wait_for_network),
    receive
        connected ->
            erlang:display(connected);
        {ok, IpInfo} ->
            erlang:display(IpInfo),
            tcp_client_start();
        disconnected ->
            erlang:display(disconnected)
    after 15000 ->
       erlang:display(wait_for_network_timeout),
       ok
   end,
    wait_for_network().

tcp_client_start() ->
    erlang:display("Connecting...\n"),
    Active = true,
    case ?GEN_TCP:connect("www.google.com", 80, [{active, Active}]) of
        {ok, Socket} ->
            erlang:display({connected, Socket}),
```
How is it intended to get a WiFi-Connection?

fadushin, 2019-09-11:
How did it fail?  Do you get as far as trying to connect?

I can get to:

```
Starting: tcp_client_esp32.beam...
---
start
I (250) wifi: wifi driver task: 3ffc3de8, prio:23, stack:3584, core=0
I (250) wifi: wifi firmware version: 9415913
I (250) wifi: config NVS flash: enabled
I (250) wifi: config nano formating: disabled
I (260) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (270) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (310) wifi: Init dynamic tx buffer num: 32
I (310) wifi: Init data frame dynamic rx buffer num: 32
I (310) wifi: Init management frame dynamic rx buffer num: 32
I (310) wifi: Init static rx buffer size: 1600
I (310) wifi: Init static rx buffer num: 10
I (320) wifi: Init dynamic rx buffer num: 32
I (320) NETWORK: starting wifi: SSID: [TheInnerSanctum], password: [XXXXXXXX].
I (400) phy: phy_version: 4008, c9ae59f, Jan 25 2019, 16:54:06, 0, 0
I (400) wifi: mode : sta (3c:71:bf:84:d9:08)
I (400) NETWORK: SYSTEM_EVENT_STA_START received.
wait_for_network
I (1130) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2110) wifi: state: init -> auth (b0)
I (2120) wifi: state: auth -> assoc (0)
I (2130) wifi: state: assoc -> run (10)
I (2150) wifi: connected with TheInnerSanctum, channel 6
I (2150) wifi: pm start, type: 1

I (2150) NETWORK: SYSTEM_EVENT_STA_CONNECTED received.
I (2710) event: sta ip: 192.168.1.236, mask: 255.255.255.0, gw: 192.168.1.1
I (2710) NETWORK: SYSTEM_EVENT_STA_GOT_IP: 192.168.1.236
connected
wait_for_network
{{192,168,1,236},{255,255,255,0},{192,168,1,1}}
[67,111,110,110,101,99,116,105,110,103,46,46,46,10]
creating TCP socketresolving to www.google.com:80 over socket fd 54
{connected,{socket,54,tcp,80,<0.5.0>,<0.1.0>,<0.6.0>}}
send: data with len: 18
<<72,84,84,80,47,49,46,48,32,50,48,48,32,79,75,13,10,68,97,116,101,58,32,87,101,100,44,32,49,49,32,83,101,112,32,50,48,49,57,32,48,49,58,49,53,58,53,48,32,71,77,84,13,10,69,120,112,105,114,101,115,58,32,45,49,13,10,67,97,99,104,101,45,67,111,110,116,114,111,108,58,32,112,114,105,118,97,116,101,44,32,109,97,120,45,97,103,101,61,48,13,10,67,111,110,116,101,110,116,45,84,121,112,101,58,32,116,101,120,116,47,104,116,109,108,59,32,99>>
```

On the ESP32, the program seems to hang waiting on data:

```
E (10380) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10380) task_wdt:  - IDLE0 (CPU 0)
E (10380) task_wdt: Tasks currently running:
E (10380) task_wdt: CPU 0: main
E (10380) task_wdt: CPU 1: IDLE1
```

My next order of business is to port this code up to the head of the current AtomVM code base, so that it can be debugged.  The branch this code is on is out of data and is not worth maintaining.  Hold tight, we'll have it working soon.

fadushin, 2019-09-15:
@flbauer That sounds like you are not authenticating to your WIFI properly.  Once the received event triggers, the network code should try to authenticate using the supplied ssid and psk.

You might also try the udp server example on the master branch, as that is better supported at this point.

fadushin, 2019-09-15:
I have rebased my `gen_tcp` branch (https://github.com/fadushin/AtomVM/tree/gen-tcp) over master, and I have ported the `generic_unix` socket driver to the updated model.

Note that the `esp32` socket driver does NOT currently work with this branch, as I still need to port the TCP client code to use the netconn API.

flbauer, 2019-09-15:
The SID  and the Password are correct. They work in plain esp-idf-Examples. 
However when I netcat to the ESP32 it never receives something.

```

 (769) NETWORK: starting wifi: SSID: [WLAN-001A4F3D9E5B], password: [XXXXXXXX].
I (869) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (869) wifi: mode : sta (c4:4f:33:11:7a:e1)
I (869) NETWORK: SYSTEM_EVENT_STA_START received.
I (1599) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2139) wifi: state: init -> auth (b0)
I (2159) wifi: state: auth -> assoc (0)
I (2169) wifi: state: assoc -> run (10)
I (2189) wifi: connected with WLAN-001A4F3D9E5B, channel 6, BW20, bssid = 9c:c7:a6:a4:84:47
I (2199) wifi: pm start, type: 1

I (2199) NETWORK: SYSTEM_EVENT_STA_CONNECTED received.
I (5149) tcpip_adapter: sta ip: 192.168.2.109, mask: 255.255.255.0, gw: 192.168.2.1
I (5149) NETWORK: SYSTEM_EVENT_STA_GOT_IP: 192.168.2.109
connected
{{192,168,2,109},{255,255,255,0},{192,168,2,1}}
Opening socket 44444 ...
<0.6.0>
Waiting to receive data...

```

fadushin, 2019-09-18:
Try sending UDP packets (e.g., via netcat -u).  You can also try the udp_client.avm example from your dev machine, but you will need to edit the file to use your IP address to send packets to and rebuild the AVM file.

fadushin, 2019-10-03:
My branch has been rebased over master, and it now works on ESP32.  https://github.com/bettio/AtomVM/pull/117  (Next, tcp server-side)

bettio, 2019-10-08:
gen_tcp has been merged in master.

closed 2019-10-08 by bettio