atomvm/AtomVM #115: esp32 - wrong include paths

flbauer, 2019-09-08:
The recent esp32-snapshots (Toolchain version: esp32-2019r1)  require 2 include-File changes in order to compile AtomVM for esp32:

**AtomVM/src/platforms/esp32/main:**

**sys.c**
```
#include <posix/sys/socket.h>
 -> #include "absolute path to $(IDF_PATH)/components/lwip/lwip/src/include/compat/posix/sys/socket.h"
```
**network_driver.c:**
```
#include <lwip/apps/sntp.h>
--> #include <apps/sntp/sntp.h>
```

There is a component.mk in the _components/libatomvm_ where a INCLUDE_FIXUP_DIR could be set. Obviously this does not affect the generation in main. The _component.mk_ file in main however is empty. Maybe one could specify the Include-Corrections there

flbauer, 2019-09-08:
For **sys.c**

```
#include <posix/sys/socket.h>
 -> #include <compat/posix/sys/socket.h>
```

works.

flbauer, 2019-09-13:
With the most recent esp-idf and xtensa-esp32-elf/esp32-2019r1-8.2.0 the include-patches I can build AtomVM. But I cannot build it with esp-idf v3.2 as it crashes. What would you recommend?

```
/home/florian/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/florian/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)

```

bettio, 2019-09-17:
I'm using this toolchain right now:

```
$ xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
```
Our CI is downloading it from here: https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz

mazorvanunu, 2020-09-13:
what the hell is going on with esp idf? why do you must break compatibility with every update??? WTF?! STOP IT, quit the job and go home! make room for people who can still salvage the situation!

bettio, 2023-06-22:
Right now we support all versions from 4.2.4 to 5.1, so I think this issue is not relevant anymore.