atomvm/AtomVM #121: Simple HTTP-Webserver

flbauer, 2019-10-24:
Hello,

I tried a little web server:

`-module(ws).
-export([start/0]).

-include("estdlib.hrl").

start() ->
    erlang:display(start),
    try
        Self = self(),
        Config = [
            {sta, [
                {ssid, "xxxx"},
                {psk,  "yyyy"},
                {connected, fun() -> Self ! connected end},
                {got_ip, fun(IpInfo) -> Self ! {ok, IpInfo} end},
                {disconnected, fun() -> Self ! disconnected end}
            ]}
        ],
        case network_fsm:start(Config) of
            ok ->
                wait_for_network();
            Error ->
                erlang:display(Error)
        end
    catch
        _:E ->
            erlang:display(E)
    end.

wait_for_network() ->
    receive
        connected ->
            erlang:display(connected),
            wait_for_network();
        {ok, IpInfo} ->
            erlang:display(IpInfo),
            http_server_start();
        disconnected ->
            erlang:display(disconnected)
    end,
    infinite_loop().

infinite_loop() ->
    ?TIMER:sleep(1000),
    infinite_loop().

http_server_start() ->
    spawn(fun () -> {ok, Sock} = ?GEN_TCP:listen(80, [{active, false}]), 
                    loop(Sock) end).

loop(Sock) ->
    {ok, Conn} = ?GEN_TCP:accept(Sock),
%%    Handler = spawn(fun () -> handle(Conn) end),
    spawn(fun () -> handle(Conn) end),
%%    ?GEN_TCP:controlling_process(Conn, Handler),
    loop(Sock).

handle(Conn) ->
    ?GEN_TCP:send(Conn, "HTTP/1.0 200 OK\nContent-Type: text\nContent-Length: 11\n\nHello World"),
    ?GEN_TCP:close(Conn).
`

It works a bit but crashes due to a missing event descriptor.
Any idea?

`connected
{{192,168,2,104},{255,255,255,0},{192,168,2,1}}
socket: init
socket: creating server tcp netconn
socket: opened event descriptor 0 in init_server_tcp_socket for netconn 0x3ffc19d4
socket: binding to IP_ADDR_ANY on port 80
socket: bound to port 80
socket: listening on port 80
socket: accepting tcp netconn
socket: Found event descriptor in socket_driver_do_accept for netconn 0x3ffc19d4
socket: accepting tcp netconn COMPLETE
socket: netconn callback, netconn: 0x3ffc19d4, evt: 0, len: 0
	event_descriptor[0]: 0x3ffc19d4
socket: accepting callback
socket: Found event descriptor in accept_callback for netconn 0x3ffc19d4
socket: netconn callback, netconn: 0x3ffc26d8, evt: 0, len: 270
	event_descriptor[0]: 0x3ffc19d4
warning: event descriptor not found
socket: event descriptor not found in socket_callback for netconn 0x3ffc26d8
abort() was called at PC 0x4014af3c on core 0
0x4014af3c: socket_callback at /home/florian/AtomVM/src/platforms/esp32/main/socket_driver.c:90
`

fadushin, 2019-11-06:
I will look into this.  I seem to remember having some issues with passive mode on ESP32 with the netconn API.  You might try using active mode, for now.

Also, a few comments.

First, please use 3 tick marks to quote code.  That way, you should not loose formatting.  E.g.,

```
-module(ws).
-export([start/0]).

-include("estdlib.hrl").

start() ->
    erlang:display(start),
    try
        Self = self(),
        Config = [
            {sta, [
                {ssid, "xxxx"},
                {psk, "yyyy"},
                {connected, fun() -> Self ! connected end},
                {got_ip, fun(IpInfo) -> Self ! {ok, IpInfo} end},
                {disconnected, fun() -> Self ! disconnected end}
            ]}
        ],
    case network_fsm:start(Config) of
        ok ->
            wait_for_network();
        Error ->
            erlang:display(Error)
    end
    catch
        _:E ->
            erlang:display(E)
    end.

wait_for_network() ->
    receive
        connected ->
            erlang:display(connected),
            wait_for_network();
        {ok, IpInfo} ->
            erlang:display(IpInfo),
            http_server_start();
        disconnected ->
            erlang:display(disconnected)
    end,
    infinite_loop().

infinite_loop() ->
    ?TIMER:sleep(1000),
    infinite_loop().

http_server_start() ->
    spawn(
        fun () -> 
            {ok, Sock} = ?GEN_TCP:listen(80, [{active, false}]),
            loop(Sock) 
        end
    ).

loop(Sock) ->
    {ok, Conn} = ?GEN_TCP:accept(Sock),
    %% Handler = spawn(fun () -> handle(Conn) end),
    spawn(fun () -> handle(Conn) end),
    %% ?GEN_TCP:controlling_process(Conn, Handler),
    loop(Sock).

handle(Conn) ->
    ?GEN_TCP:send(Conn, "HTTP/1.0 200 OK\nContent-Type: text\nContent-Length: 11\n\nHello World"),
    ?GEN_TCP:close(Conn).
```

Next, I would say I would recommend that in your specific program, you spawn a proc with an accept, instead of blocking in accept.  At some point, you will need to read the request, and you don't want the process of reading to block the next accept (unless you are trying to implement a single threaded server, which sometimes makes sense, but which is a problem with browsers which run requests in parallel for resources on the same node).  But that is separate from the file descriptor issue you see on ESP32, which I will look into.

Finally, it helps to verify that this does not have the same issues on UNIX, as the drivers are quite different, and it can help us pinpoint whether the issue is in driver code, or in the VM.

Please beware that this code is very very preliminary, and we are working to correct it as soon as possible.  But it's people like you who try the code and file bugs that makes this possible.  Thank you!!

bettio, 2019-11-12:
I wrote a toy http server, it might be useful for you :)
@flbauer please take a look to: https://github.com/bettio/AtomVM/blob/master/examples/erlang/hello_world_server.erl and https://github.com/bettio/AtomVM/blob/master/libs/eavmlib/http_server.erl
You might find some useful information :)

bettio, 2019-11-12:
I also applied the following workaround, so it stopped crashing:
```
diff --cc src/platforms/esp32/main/socket_driver.c
index 4ef56af,4ef56af..d3b868c
--- a/src/platforms/esp32/main/socket_driver.c
+++ b/src/platforms/esp32/main/socket_driver.c
@@@ -101,8 -101,8 +101,10 @@@ void socket_callback(struct netconn *ne
      // print_event_descriptors();
      int event_descriptor = find_event_descriptor(netconn);
      if (UNLIKELY(event_descriptor < 0)) {
--        TRACE("socket: event descriptor not found in socket_callback for netconn 0x%lx\n", (unsigned long) netconn);
--        abort();
++        fprintf(stderr, "failed find event descriptor\n");
++        return;
++        //TRACE("socket: event descriptor not found in socket_callback for netconn 0x%lx\n", (unsigned long) netconn);
++        //abort();
      }
  
      int result = xQueueSend(event_queue, &event_descriptor, 0);
```

closed 2019-12-30 by bettio

flbauer, 2020-03-11:
> I wrote a toy http server, it might be useful for you :)
> 
> @flbauer please take a look to: https://github.com/bettio/AtomVM/blob/master/examples/erlang/hello_world_server.erl and https://github.com/bettio/AtomVM/blob/master/libs/eavmlib/http_server.erl
> 
> Let me know if you find it helpful :)

Hi 
I tried your example and it works well.
Thank you!

commit f6dc6467 (2019-12-30): ESP32: revamp socket implementation

Previous implementation had several bugs.
Write a new implementation which is based on the new scheduler code and
it doesn't have all the issues it used to have.
Fixes #121.
commit 857851a5 (2019-12-30): ESP32: revamp sockets

Previous implementation had several bugs.
Write a new implementation which is based on the new scheduler code and
it doesn't have all the issues it used to have.
Fixes #121.