atomvm/AtomVM #136: Make build fails even though zlib1-dev installed

NewBermuda, 2019-12-12:
When making AtomVM on Ubuntu I'm getting the following error messages:

/home/melvin/AtomVM/tools/packbeam/packbeam.c:396:5: error: unknown type name ‘z_stream’
     z_stream infstream;
     ^
/home/melvin/AtomVM/tools/packbeam/packbeam.c:397:14: error: request for member ‘zalloc’ in something not a structure or union
     infstream.zalloc = Z_NULL;
              ^
/home/melvin/AtomVM/tools/packbeam/packbeam.c:397:24: error: ‘Z_NULL’ undeclared (first use in this function)
     infstream.zalloc = Z_NULL;
                        ^
/home/melvin/AtomVM/tools/packbeam/packbeam.c:397:24: note: each undeclared identifier is reported only once for each function it appears in
/home/melvin/AtomVM/tools/packbeam/packbeam.c:398:14: error: request for member ‘zfree’ in something not a structure or union
     infstream.zfree = Z_NULL;
and so on.

This probaly has to do with the zlib devel files not being correctly recognized when making.
I installed zlib1g-dev via apt-get install if that is what was meant by 'zlib devel files'
Do you have any idea what I'm missing here?
Help would be greatly appreciated

bettio, 2019-12-17:
Which Ubuntu version are you using?
Can you provide cmake output?

NewBermuda, 2019-12-18:
I'm sorry, actually my installation was faulty. Works nicely now

closed 2019-12-18 by NewBermuda