atomvm/AtomVM #150: Improve Elixir support

bettio, 2020-02-01:
Following modules are missing and they should be implemented:
- Enum
- GenServer
- Process
- Tuple
- List
- String
- Integer
- IO

OvermindDL1, 2020-02-03:
Those aren't beam-style stdlib things, Elixir makes it's own beam files for those that get included with elixir-made releases and they are just stock normal beam files.  By including those you will create a conflict.

bettio, 2020-02-12:
There are some point worth to be mentioned:
1. we are not relying on Elixir releases, we would like some mix tasks in near future, but I don't think we are going to use releases.
2. Elixir standard modules are huge, we need a subset of the supported functions so we can save precious flash space. This means we need to strip out features/functions.
3. We are not able to run Elixir modules out of the box, there are still some missing erlang standard functions and some opcodes are still missing. We should start with smaller modules with usable functions.

For this specific reasons we might need some custom Elixir modules, anyway I would like to discuss how to achieve our goals with an approach that works well with Elixir way of doing things.

vctrferreira, 2021-03-19:
Hello, what is the current status from this roadmap? I'm getting starts with the project using Elixir environment for now I'm just planning my project but I need to know what is the current status to may trying to start to contribute.

uDude, 2021-03-19:
I thnks first thought be knwo feature of erlang roadmap.

Then someone (maybe ask Jose if he is interest in do support work for
eElixir on this eErlang).

Ist idea gut, aber es ist eine grande project, by self.

uDude, 2021-03-19:
BTW love this projekt

bettio, 2022-03-30:
Sorry for not updating this issue, here are some news:
Following modules from the previous list has been implemented:
- Enum
- Process
- Tuple
- List

Also a number of other modules have been added like Access, Kernel, Code, Module, etc...

I'm going to track each module using a separate issue here:
- https://github.com/atomvm/AtomVM/issues/265
- https://github.com/atomvm/AtomVM/issues/266
- https://github.com/atomvm/AtomVM/issues/266
- https://github.com/atomvm/AtomVM/issues/267
- https://github.com/atomvm/AtomVM/issues/268

bettio, 2022-03-30:
Closing this issue, please open any new issue or comment to above new issues for any further discussion/update.

closed 2022-03-30 by bettio