atomvm/AtomVM #153: AtomVM + Elixir + Contexterlang

jorgeehernandez, 2020-03-05:
Thanks for the work on this project!  

Currently I'm trying to find out if I can use AtomVm on my master tesis . One of the aim of the tesis is to run  http://www.guidosalvaneschi.com/wp/contexterlang/ using elixir. Contexterlang relies on Erlang concurrency model to create context aware agents.  The idea is to use elixir with AtomVM +  Contexterlang to create IoT agents that runs on devices esp32 and stm32. 

- I would like to know  what do you think about this possible use of AtomVM? 

- Is AtomVM (actual state) able to run this kind of COP language (Contexterlang) ?

Thanks

arpunk, 2020-03-05:
AtomVM implements a subset of Erlang and there is currently no OTP support. That means most of the libraries that `contexterlang` uses are not avialable and must be backported. `contexterlang` seems to use `ets`, `rpc`, `code` and other modules not available in AtomVM.

bettio, 2020-03-29:
Did you make any progress? Do you have any feedback/information you would like to share?