atomvm/AtomVM #161: Recipe for component fails

FlashPackets, 2020-05-06:
From a fresh install of esp_idf 4.2 I did the following
git fetch
git describe --tags --dirty
v4.2-dev-1395-gce2a99dc2
git checkout v3.2
...
git submodule update --init  --recursive
...
git describe --tags --dirty
# Then in the ../AtomVM/src/platforms/esp32 
make      # and got 

AR build/bootloader/main/libmain.a
LD build/bootloader/bootloader.elf
esptool.py v2.6
Python requirements from ~/esp/esp-idf/requirements.txt are satisfied.
Building partitions from ~/esp/3rdParty/AtomVM/src/platforms/esp32/partitions.csv...
CC build/app_trace/app_trace_util.o
CC build/app_trace/host_file_io.o
CC build/app_trace/app_trace.o
CC build/app_trace/gcov/gcov_rtio.o
AR build/app_trace/libapp_trace.a
CC build/app_update/esp_ota_ops.o
AR build/app_update/libapp_update.a
CXX build/asio/asio/asio/src/asio.o
In file included from /home/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /home/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/ext/string_conversions.h:41,
                 from /home/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/bits/basic_string.h:6391,
                 from /home//.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/string:52,
                 from /hdd/home/esp/esp-idf/components/asio/asio/asio/include/asio/impl/error.ipp:19,
                 from /hdd/home/esp/esp-idf/components/asio/asio/asio/include/asio/impl/src.hpp:22,
                 from /hdd/home/esp/esp-idf/components/asio/asio/asio/src/asio.cpp:11:
/home/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
/hdd/home/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 1
/hdd/home/esp/esp-idf/make/project.mk:530: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2

Is not v3.2  the correct version of idf?
The build has gotten all the way to the linker, .... so close
How doesone get to the finish line?

FlashPackets, 2020-05-07:
Problem fixed.
deleting the existing build,
mkdir build
cd build
cmake ..
make 

fixed the issue!

fadushin, 2020-05-07:
Are you trying to target Unix (Linux, MacOS, or FreeBSD), or are you trying to target ESP32?

Your cmake; make steps will build AtomVM for UNIX(like) systems.  If you want to target ESP32, you need to use IDF SDK 3.3.  4.x is not yet supported.

closed 2022-03-28 by bettio