atomvm/AtomVM #172: Out of order messages hang on receive

fadushin, 2020-07-04:
The following code works fine on OTP but appears to hang on AtomVM
```
-module(test).

-export([start/0]).

start() ->
    Self = self(),
    spawn(fun() -> Self ! two end),
    spawn(fun() -> receive after 1000 -> ok end, Self ! one end),
    receive
        one ->
            erlang:display(one)
    end,
    receive
        two ->
            erlang:display(two)
    end.
```

Under OTP:

```
Erlang/OTP 22 [erts-10.7.1] [source] [64-bit] [smp:8:8] [ds:8:8:10] [async-threads:1] [hipe] [dtrace]

Eshell V10.7.1  (abort with ^G)
1> test:start().
one
two
true
```

Under AtomVM,

fadushin, 2020-12-12:
See https://github.com/bettio/AtomVM/pull/188

bettio, 2021-01-24:
Fixed  with PR #188.

closed 2021-01-24 by bettio