atomvm/AtomVM #177: Problems with the is_float BIF

polvalente, 2020-09-30:
I tried running the same commands defined in the CI workflow to compile and test the system, to ensure the code was running correctly in my machine.

However, the code always fails with `Undecoded opcode: 94` when the `is_float/1` BIF is called. If I remove all mentions to it from the tests, the workflow succeeds as expected.

Any insights on what could be going on?

fadushin, 2020-09-30:
OS and version?  OTP version?

polvalente, 2020-09-30:
Registering the updates from Slack:
- @fadushin commented that he set some flags with CMAKE
- Afterwards, I tried to run the following sequence of commands from the clean `master` branch, which finally resulted in a success:

```bash
mkdir -p build
cd build
cmake -DAVM_DISABLE_FP=on ..
make
./tests/test-erlang
```

As @fadushin explained, the flag disabled floating point support, but in my opinion, it's nice to have a clean test run for development.

closed 2020-09-30 by polvalente