atomvm/AtomVM #180: error running .avm files compiled by Elixir

mhama, 2020-10-02:
Hi, I tried to run elixir scripts with AtomVM, but failed.

After cmake and make, I tried to run Linux version of AtomVM.
Running erlang/hello_world.avm was successful, but running elixir/HelloWorld.avm failed with `WARNING: Invalid port command.  Unable to send reply` error.
I also tried ESP32 version and the elixir script stops without any console output, whereas erlang scripts seems ok.

Are there any solutions?

Erlang avm:
```sh
/home/AtomVM/build# src/AtomVM examples/erlang/hello_world.avm
Hello World
Return value: ok
```

Elixir avm:
```sh
/home/AtomVM/build# src/AtomVM examples/elixir/HelloWorld.avme
WARNING: Invalid port command.  Unable to send reply
```

Environment: Ubuntu 18.04 on Windows WSL2 (Ubuntu 18.04) docker
based on /docker/Dockerfile and 
added erlang/elixir like this:

```
RUN wget https://packages.erlang-solutions.com/erlang-solutions_2.0_all.deb && \
dpkg -i erlang-solutions_2.0_all.deb && \
apt-get update

RUN apt-get install -y esl-erlang=1:22.3.4.9-1
RUN apt-get install -y elixir zlib1g-dev
```

mhama, 2020-10-04:
After that, I found that the elixir/esp32/Blink sample run successfully on ESP32, but HelloWorld sample does not run well still.
The HelloWorld sample raise `WARNING: Invalid port command.  Unable to send reply` error on Linux, and the same script stops without any log on ESP32.

log on ESP32:

```
...
Found AVM partition: size: 1048576, address: 0x210000
Booting file mapped at: 0x3f430000, size: 1048576
Found AVM partition: size: 1048576, address: 0x110000
Starting: Elixir.HelloWorld.beam...
---
```

UncleGrumpy, 2022-04-30:
This issue is probably being caused by issue #272. The Elixir beams are not being packed into atomvmlib.avm like they should be.

UncleGrumpy, 2022-11-06:
This is still an problem and is related to Issue #377.

UncleGrumpy, 2022-11-23:
This issue should be resolved after the merge of #367, an should be safe to close.

closed 2023-06-08 by UncleGrumpy