atomvm/AtomVM #201: is AtomVM still alive?

elsbiet, 2021-04-10:
this is not an issue but a question. because i do not see much activity: is this project dead or alive?

thanks in advance for your answers, e.

bettio, 2021-04-10:
Alive: https://github.com/bettio/AtomVM/commits/master

@fadushin has recently did a great work by implementing maps (which enables us full Elixir power), while I'm finishing adding [support to supervisors](https://github.com/bettio/AtomVM/commit/96e2ffc874096d2c9192f3a7114d91e45989d04c) ;)
We are close to tagging the first release, since we have support for all the most important erlang/elixir functions.

bettio, 2021-04-10:
Also real time chat has been moved to discord: https://discord.gg/n8V7t7rw

elsbiet, 2021-04-10:
i'm glad to read your words.

thanks for the info, e.

closed 2021-04-10 by elsbiet

elsbiet, 2021-04-25:
thanks for your fast answer bettio,

currently i try to start a little project  using atomvm but i find it hard to get restarted.

i cloned your repository (git clone https://github.com/bettio/AtomVM.git) and compiled it but i an quite unhappy.

eg. atomvm seems to be unable to start proceses. i try to run

-module(p).
 
-export([start/0, ping/2, pong/0]).

ping(0, Pong_PID) ->
    Pong_PID ! finished,
    io:format("ping finished~n", []);

ping(N, Pong_PID) ->
    Pong_PID ! {ping, self()},
    receive
        pong ->
            io:format("Ping received pong~n", [])
    end,
    ping(N - 1, Pong_PID).

pong() ->
    receive
        finished ->
            io:format("Pong finished~n", []);
        {ping, Ping_PID} ->
            io:format("Pong received ping~n", []),
            Ping_PID ! pong,
            pong()
    end.

start() ->
    io:format("starting pong..~n"),
    Pong_PID = spawn(p, pong, []),
    io:format("starting ping..~n"),
    spawn(p, ping, [3, Pong_PID]).

this little pgm just outputs:

..
Starting: p.beam...
---
starting pong..
starting ping..
AtomVM finished with return value = <0.3.0>
going to sleep forever..

is there anything i might have missed?

thanks for your answer.

gustav

fadushin, 2021-04-25:
Try adding

```
receive X -> X end
```

to the end of your program.

What is happening is that your program is terminating immediately.

You may also want to use `?MODULE` in place of `p` in your program.

elsbiet, 2021-04-26:
thanks,

this was the missing part - may fault.

thanks again.

gustav