atomvm/AtomVM #205: out of memory

elsbiet, 2021-04-30:
i know, it's not a typical use case..

running 

-module(nx).

-export([start/0]).

-define(LIMIT, 10000).

repeat(_,N) when is_integer(N), N =< -1 -> ok;
repeat(_,0) -> ok;
repeat(F,N) -> F(N), repeat(F, N - 1).

start() ->
    repeat(
      fun(N) ->
              L = seq(1, ?LIMIT - N),
              R = lists:reverse(L),
              io:format("N=~p~n", [N]),
              receive
              after 5 ->
                        L = lists:reverse(R),
                        ok
              end
      end,
      ?LIMIT)
    .

seq(First, Last)
  when is_integer(First), is_integer(Last), First-1 =< Last ->
    seq_loop(Last-First+1, Last, []);

seq(_, Last)
  when is_integer(Last), Last =< 0 ->
    [].

seq_loop(N, X, L) when N >= 4 ->
    seq_loop(N-4, X-4, [X-3,X-2,X-1,X|L]);
seq_loop(N, X, L) when N >= 2 ->
    seq_loop(N-2, X-2, [X-1,X|L]);
seq_loop(1, X, L) ->
    [X|L];
seq_loop(0, _, L) ->
    L.

on an esp32 results in: [1,..,7060,7061,7062,7063,7064,7065,7066,7067,7068,7069,7070,7071,7072,7073,7074,7075,7076,7077,7078,7079,7080,7081,7082,7083,7084,7085,7086,7087]
[]
#CP<module: 0, label: 5, offset: 21>
2913
#Fun<erl_eval.15.13958363296>
#CP<module: 0, label: 25, offset: 0>

Registers
----------
x[0]: error
x[1]: out_of_memory
x[2]: []
x[3]: []
x[4]: []
x[5]: []
x[6]: []
x[7]: []
x[8]: []
x[9]: []
x[10]: []
x[11]: []
x[12]: []
x[13]: []
x[14]: []
x[15]: []

Mailbox
--------

**End Of Crash Report**
AtomVM finished with return value = error
going to sleep forever..

running the same pgm in an atomvm on linux with restricted memory (ulimit -Sv 9940 before starting atomvm) will be out of memory at N=8803 on my machine.

fadushin, 2021-05-25:
If I understand your code, you are creating a list of up to say 8k integers, but in general you will have up to three copies of these in memory at any given time (the list you created via `seq`, the reversed list, and then another reverse of the reverse.  Each integer occupies a word of memory (4 or 8 bytes, depending on the architecture), and each list cons occupies 2 words.  So I suspect the process heap is around 3*(N*3)*word_size bytes.  For 8k entries on an ESP32, that should be around 288k, which sounds about right -- you start out with around that much free memory, assuming you are not using Wifi or any networking.

Have a look at the erlang:process_info and erlang:system_info functions.  You might get some useful stats out of the VM in your experiment.

bettio, 2022-03-28:
I'm closing this issue since it is quite tied to intrinsic hardware limitations.

closed 2022-03-28 by bettio