atomvm/AtomVM #206: Make error on OSX

wwhai, 2021-05-07:
```
2 warnings generated.
[  6%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/port.c.o
[  6%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/refc_binary.c.o
[  6%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/scheduler.c.o
[  7%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/term.c.o
[  7%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/timer_wheel.c.o
[  7%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/valueshashtable.c.o
[  7%] Linking C static library liblibAtomVM.a
[  7%] Built target libAtomVM
Scanning dependencies of target AtomVM
[  7%] Building C object src/CMakeFiles/AtomVM.dir/main.c.o
[  7%] Linking C executable AtomVM
ld: cannot link directly with dylib/framework, your binary is not an allowed client of /usr/lib/libcrypto.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/AtomVM] Error 1
make[1]: *** [src/CMakeFiles/AtomVM.dir/all] Error 2
make: *** [all] Error 2
```
My OSX version：
[image]

wwhai, 2021-05-07:
> ```
> 2 warnings generated.
> [  6%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/port.c.o
> [  6%] Building C object src/libAtomVM/CMakeFiles/libAtomVM.dir/refc_binary.c.o
> [  6%] Building C [...]

## I have resolved by update OSX to big sur 11.3.1 (20E241)!

fadushin, 2021-05-24:
I use openssl from brew.  E.g.,

```
shell$ brew install openssl
...
```

And then

```
shell$ cmake -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl -DOPENSSL_LIBRARIES=/usr/local/opt/openssl/lib ..
```

closed 2021-05-28 by wwhai