atomvm/AtomVM #207: error: invalid AVM Pack

FlashPackets, 2021-05-13:
I'm so... excited. 

AtomVM now compiles under the the latest 4.x release of the esp_idf tools. This is really great news and should really light the fires of progress. Well done! 
Following the README.ESP32.Md instructions ..

The compilation process was trouble free. However, when I hooked up the monitor I get the following curious behavior.

ELF file SHA256: bd6402737071d36e

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6928
ho 0 tail 12 room 4
load:0x40078000,len:13852
load:0x40080400,len:4548
entry 0x400806ec
I (31) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (31) boot: compile time 14:28:30
I (31) boot: chip revision: 0
I (34) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 4MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (86) boot:  3 lib.avm          RF data          01 01 00110000 00100000
I (94) boot:  4 main.avm         RF data          01 01 00210000 00100000
I (101) boot: End of partition table
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1e418 (123928) map
I (161) esp_image: segment 1: paddr=0x0002e440 vaddr=0x3ffb0000 size=0x01bd8 (  7128) load
I (165) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0xba96c (764268) map
0x400d0020: _stext at ??:?

I (457) esp_image: segment 3: paddr=0x000ea994 vaddr=0x3ffb1bd8 size=0x02a8c ( 10892) load
I (461) esp_image: segment 4: paddr=0x000ed428 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/flash/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (464) esp_image: segment 5: paddr=0x000ed830 vaddr=0x40080400 size=0x18edc (102108) load
I (517) esp_image: segment 6: paddr=0x00106714 vaddr=0x400c0000 size=0x00064 (   100) load
I (531) boot: Loaded app from partition at offset 0x10000
I (531) boot: Disabling RNG early entropy source...
I (532) cpu_start: Pro cpu up.
I (535) cpu_start: Application information:
I (540) cpu_start: Project name:     atomvvm-esp32
I (545) cpu_start: App version:      32acfee-dirty
I (551) cpu_start: Compile time:     May 12 2021 14:28:44
I (557) cpu_start: ELF file SHA256:  bd6402737071d36e...
I (563) cpu_start: ESP-IDF:          v4.2-dirty
I (568) cpu_start: Starting app cpu, entry point is 0x40081528
0x40081528: call_start_cpu1 at /home/flash/esp/esp-idf/components/esp32/cpu_start.c:287

I (560) cpu_start: App cpu up.
I (579) heap_init: Initializing. RAM available for dynamic allocation:
I (586) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (592) heap_init: At 3FFB8C88 len 00027378 (156 KiB): DRAM
I (598) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (604) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (611) heap_init: At 400992DC len 00006D24 (27 KiB): IRAM
I (617) cpu_start: Pro cpu start user code
I (635) spi_flash: detected chip: gd
I (636) spi_flash: flash io: dio
I (636) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Found AVM partition: size: 1048576, address: 0x210000
Booting file mapped at: 0x3f420000, size: 1048576
error: invalid AVM Pack     <===================================================================

abort() was called at PC 0x400d2f63 on core 0
0x400d2f63: app_main at /home/flash/Documents/PlatformIO/Projects/AtomVM/src/platforms/esp32/main/main.c:73

Backtrace:0x40089a8b:0x3ffbaa60 0x4008a129:0x3ffbaa80 0x40090cee:0x3ffbaaa0 0x400d2f63:0x3ffbab10 0x400d14c6:0x3ffbab40 0x4008be2d:0x3ffbab70
0x40089a8b: panic_abort at /home/flash/esp/esp-idf/components/esp_system/panic.c:330

0x4008a129: esp_system_abort at /home/flash/esp/esp-idf/components/esp_system/system_api.c:106

0x40090cee: abort at /home/flash/esp/esp-idf/components/newlib/abort.c:46

0x400d2f63: app_main at /home/flash/Documents/PlatformIO/Projects/AtomVM/src/platforms/esp32/main/main.c:73

0x400d14c6: main_task at /home/flash/esp/esp-idf/components/esp32/cpu_start.c:600 (discriminator 2)

0x4008be2d: vPortTaskWrapper at /home/flash/esp/esp-idf/components/freertos/xtensa/port.c:143

What should be done to fix the invalid AVM Pack ?

fadushin, 2021-05-25:
Can you provide a little bit more information about what steps you performed, so that we can try to recreate?  Thanks!

bettio, 2022-03-28:
There is no way to fix this issue with so few information, so I'm closing it. Feel free to open a more detailed issue anytime.

closed 2022-03-28 by bettio