atomvm/AtomVM #223: Cannot make for ESP32

dmbr0, 2022-01-10:
cd AtomVM/src/platforms/esp32

➜  esp32 git:(master) ✗ make
Makefile:3: /home/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target '/home/esp/esp-idf/make/project.mk'.  Stop.

Am I missing something? I am able to build and run the hello world for esp-idf and I know my path is correct.

tlack, 2022-01-19:
What version of ESP-IDF are you using? I've had good luck with ESP-IDF v4.2.2.

Also, please confirm what version of the OTP (Erlang) platform you have installed. I'm using 22.3.4.23 at the moment (via asdf) and it's working well.

UncleGrumpy, 2022-01-20:
It looks to me like your IDF_PATH is set wrong. right now it is IDF_PATH=/home/esp/esp-idf/
I believe what you want is to set it to IDF_PATH=$HOME/esp/esp-idf.

UncleGrumpy, 2022-03-23:
It looks like you are building with ESP-IDF 4.3 or 4.4. Unfortunately until we update our build system AtomVM will only compile with Version 4.2.x or previous of the ESP-IDF. I would recommend version 4.2.3 until we have resolved this issue.

gabrielmancini, 2022-03-23:
i change to 4.2.2 i have an update there

UncleGrumpy, 2022-03-23:
Since you switched versions you may have to delete the entire “build” directory in AtomVM/src/platforms/esp32 and try the build again.

bettio, 2022-03-28:
@gabrielmancini let us know if last @UncleGrumpy advice did help for your issue.
Meanwhile I'm going to reference https://github.com/atomvm/AtomVM/issues/246 which is related to some of the esp-idf version issues reported here.

dmbr0, 2022-05-05:
I ended up solving this by using a previous version of esp-idf. I had installed the newest version before I saw which version was required.

closed 2022-05-05 by dmbr0