atomvm/AtomVM #250: Move atomvm_lib ADC Nif to AtomVM

fadushin, 2022-03-29:
The atomvm_lib repo contains an integration with the ADC API in the IDF SDK.  Since this represents core functionality of the ESP32, this code should be moved into the AtomVM esp32 platform as a component.

biennguyen94, 2023-07-14:
Do you have an estimated plan to implement this?
It seems latest atomvm_adc is not compatible with latest AtomVM, I can build AtomVM image successfully but a crash happened when flashing the .avm to esp32.
I am now using AtomVM 0.5.0 and old atomvm_adc version (around Mar 2022), I can see that adc is working well.

UncleGrumpy, 2023-07-14:
The atomvm_adc driver should work with esp-idf 4.x versions, but currently will not compile using esp-idf 5.x versions. The APIs completely changed and support for IDF 5.x is taking longer than anticipated, but should be available soon(ish)... ;)

biennguyen94, 2023-07-15:
I built AtomVM image including ADC on ubuntu 22.04, Erlang OTP 25, esp-idf 4.4, latest AtomVM, latest atomvm_adc, then I flashed this image to ESP32, then I flashed the application to ESP32 but error happened as below:
[image]

UncleGrumpy, 2023-07-15:
Since we can't merge the atomvm_adc component into AtomVM until it is updated to compile with ESP-IDF 5.x versions, would you mind opening an issue in the atomvm_adc repo so we can help you sort out this problem there?

biennguyen94, 2023-07-15:
thanks @UncleGrumpy, I have created a new issue in the atomvm_adc repo.

pguyot, 2023-07-15:
So eventually we decided to put adc into the VM? Doesn't it imply adding an additional component dependency?

UncleGrumpy, 2023-07-15:
No, the only components that the ADC driver uses are included in the esp-idf. They only thing preventing moving atomvm_adc to AtomVM core is that it currently doesn't build with esp-idf 5.x due to complete breakage of the API in version 5 (there is literally nothing in common between 4.x and 5.x - all the way down to different data types). I have already started work on IDF 5 support, so hopefully that won't take too much longer.

commit 87716e82 (2024-10-03): Merge pull request #1264 from UncleGrumpy/esp32_adc

Introduce ESP32 ADC driver

This set of changes adds support for the ESP32 ADC peripheral, with
documentation and examples.

Closes #250

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #1264 (2024-10-03): Introduce ESP32 ADC driver
This set of changes adds support for the ESP32 ADC peripheral, with documentation and examples.

Closes #250

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2024-10-15 by bettio