atomvm/AtomVM #299: hello_world.avm Undecoded opcode: 175

taguniversalmachine, 2022-04-22:
Building from commit fc5778c618acde0a684804cb74ea4ffd53275823

On a MacBook with OS X 11.6.2 (20G314)

CMake build with a couple of warnings, then:

[100%] Built target MultiBlink_main
[100%] Packing runnable MultiBlink.avm
[100%] Built target MultiBlink

eflores@orion build % ./src/AtomVM ./examples/erlang/hello_world.avm
Undecoded opcode: 175
zsh: abort      ./src/AtomVM ./examples/erlang/hello_world.avm

Not sure if I configured something wrong on my build environment or how to debug this.

fadushin, 2022-04-23:
It looks like you have compiled the hello_world application with OTP 24 or later.  Opcode 175 appears to have been introduced in OTP-24.  We currently only support OTP versions 21-23.

https://doc.atomvm.net/getting-started-guide.html

Also, you might be slightly better off running the AtomVM examples, since they do not require a full build of the Erlang VM.  You can download the AtomVM 0.5.0 image here:

https://github.com/atomvm/AtomVM/releases/download/v0.5.0/AtomVM-v0.5.0-ESP32.img

The example programs are here:

https://github.com/atomvm/atomvm_examples/

Can you please rebuild using OTP-23, and let us know if that works?

taguniversalmachine, 2022-04-23:
Tried with OTP 23 and 22 but still get the error..

[100%] Built target MultiBlink_main
[100%] Packing runnable MultiBlink.avm
[100%] Built target MultiBlink
eflores@orion build % ./src/AtomVM ./examples/erlang/hello_world.avm
Undecoded opcode: 175
zsh: abort      ./src/AtomVM ./examples/erlang/hello_world.avm
eflores@orion build % elixir -v
Erlang/OTP 24 [erts-12.0.3] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [jit] [dtrace]

Elixir 1.11.0 (compiled with Erlang/OTP 23)

./src/AtomVM ./examples/erlang/hello_world.avm
Undecoded opcode: 175
zsh: abort      ./src/AtomVM ./examples/erlang/hello_world.avm
eflores@orion build % elixir -v
Erlang/OTP 24 [erts-12.0.3] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [jit] [dtrace]

Elixir 1.12.2 (compiled with Erlang/OTP 22)

UncleGrumpy, 2022-04-23:
You can check by running `asdf current` and change it by running `asdf global erlang 22.3.4.24` and `asdf global elixir 1.12-otp-22` for example. It looks like it wasn't changed because the `elixir -v` for both runs still show OTP 24.

taguniversalmachine, 2022-04-23:
Ok I have OTP 22 everywhere now but still seems to fail:
..
[100%] Packing runnable MultiBlink.avm
[100%] Built target MultiBlink
%  ./src/AtomVM ./examples/erlang/hello_world.avm
Undecoded opcode: 175
zsh: abort      ./src/AtomVM ./examples/erlang/hello_world.avm
eflores@orion build % elixir -v
Erlang/OTP 22 [erts-10.7.2.16] [source] [64-bit] [smp:4:4] [ds:4:4:10] [async-threads:1] [hipe]

Elixir 1.12.2 (compiled with Erlang/OTP 22)

asdf current erlang
erlang          22.3.4.24

UncleGrumpy, 2022-04-23:
You might need to delete the build directory and start with a clean one, because that opcode: 175 does not exist until OTP 24. Also I just realized it does not appear in the README yet but we have much more project information available at https://atomvm.net/ where you can find a link to much more extensive documentation, and contact links including "support" chat on Telegram.

taguniversalmachine, 2022-04-23:
Awesome that was it thanks for your help and I'll check out that info.

Hello World
Return value: ok

closed 2022-04-23 by taguniversalmachine