atomvm/AtomVM #339: build for exp32

gabrielmancini, 2022-09-12:
when i build the AtomVM esp32

i got this:
```
make[1]: cmake: Permission denied
make[1]: *** [/home/barba/Development/AtomVM/src/platforms/esp32/components/libatomvm/component.mk:52: Makefile] Error 127
make: *** [/home/barba/esp/esp-idf/make/project.mk:649: component-libatomvm-build] Error 2
make: *** Waiting for unfinished jobs....
```

there is some stable branch?

UncleGrumpy, 2022-09-12:
Sorry to hear you are experiencing problems... It looks like you are the victim of a rare bug we are trying to track down.  We still are not certain why it only exhibits itself on some systems but not others, but I have a work around you can try.

If you `cd` into your esp-idf installation and edit the `CMake` entry in tools/tools.json to:
```
...
      "description": "CMake build system",
      "export_paths": [
        [
          "bin"
        ]
      ],
      "export_vars": {},
      "info_url": "https://github.com/Kitware/CMake",
      "install": "always",
      "license": "BSD-3-Clause",
      "name": "cmake",
...
```

The line that need to be changed is the "install" line, it will say "on-demand", but if you change that to "always", save the file and then run `./install.sh` again you should see the correct version of cmake being installed, and then after sourcing export.sh again (`. ./export.sh`) again you should have no problems building AtomVM.
If that does not work please report back and we will see if we can help get to the bottom of it.

gabrielmancini, 2022-09-12:
oh thanks @UncleGrumpy now build succeed.
but next when i flash the esp nor work properly,
there is way to easy flash the esp32 with AtomVM and the standard library?
i try using the img from download page, but when i try to run the hello_world example its breaks beause it dont find the IO module

UncleGrumpy, 2022-09-13:
Glad to here that the workaround took care of the build problem! Since you found us in the [Telegram AtomVM Chat](https://t.me/atomvm) we can sort out your next issue with you there.

Until the true source of this bug has been found, can you change the title to something like "ESP32 Build Failure: cmake: Permission denied (Makefile Error 127)" that gives a little more description of the error? That way anyone else having the same problem can recognize the issue and find this temporary workaround.

bettio, 2023-06-22:
I think things changed quite a lot and this issue doesn't happen anymore.

closed 2023-06-22 by bettio