atomvm/AtomVM #377: Elixir Console driver broken.

UncleGrumpy, 2022-11-02:
The elixir `Console` driver does not output anything to the console. Attached is a short example with the actual console output from an esp32 displayed below.
[image]

UncleGrumpy, 2022-11-02:
Here is a better example, the Console.puts/1 never returns, as you can see the `:ok` is never reached.

[image]

fadushin, 2022-11-03:
I wonder if `:console.flush()` would work?  (Or your implementation here: https://github.com/atomvm/AtomVM/pull/367/files#diff-0ec03f7b1f9346acb32e975541e1c6c620e0bd87a614cfc09ee02b49d64398e0R94)

UncleGrumpy, 2022-11-06:
This looks like is is also related to Issue #180... 
Both issues can be closed by merging PR #367.

UncleGrumpy, 2022-11-13:
Fixed in merged PR #367.

UncleGrumpy, 2022-11-13:
Fixed in PR #367.

UncleGrumpy, 2022-11-13:
Fixed in PR #367.

closed 2022-11-13 by UncleGrumpy