atomvm/AtomVM #470: API: loading BEAM and AVM from filesystem, flash partitions, etc

bettio, 2023-01-22:
It should be possible somehow to load BEAM and AVM files from filesystem (hence SDs also), raw flash partitions or whatever.
A portable and uniform across platforms approach should be found.
Erlang has already some functions such as: `code:load_abs/1` that take a file path.

Open points:
How this semantic can be extended on other platforms, and in order to support RAW flash partitions such as on esp32?

bettio, 2023-01-22:
Idea:
ESP32:
- code:load_abs("/foo") -> regular ESP32 VFS
- code:load_abs("/dev/partitions/PART_NAME") -> we override the regular VFS and we get raw access to partition

fadushin, 2023-01-22:
What would be an example of `PART_NAME`?

UncleGrumpy, 2023-01-22:
My feeling at thew moment is that the first option is preferable:

- code:load_abs("/foo") -> regular ESP32 VFS

The mount point can be given an arbitrary name (PATH) by the user when it is created. This would make for easy portability with the unix_port, where the user will want to specify an arbitrary file path. This also leave the flexibility for porting to other MCU platforms that may have a specific PATH or naming scheme that must be used to access SD-Card or flash filesystems.

bettio, 2023-01-22:
> What would be an example of `PART_NAME`?

Any partition name, such as `main.avm` therefore `/dev/partitions/main.avm` or something like that.

bettio, 2023-01-22:
x

> My feeling at thew moment is that the first option is preferable:
> 
>     * code:load_abs("/foo") -> regular ESP32 VFS
> 
> 
> The mount point can be given an arbitrary name (PATH) by the user when it is created. This would make for easy portability with the unix_port, where the user will want to specify an arbitrary file path. This also leave the flexibility for porting to other MCU platforms that may have a specific PATH or naming scheme that must be used to access SD-Card or flash filesystems.
> 
> I think it would be more beneficial to the end user to use the `fat` filesystem rather than raw sd-card access. This would give end users a lot more flexibility, as the files on the sd-card could be updated by any system with a card reader/writer.

Let me explain me better: `code:load_abs("/foo")` and `code:load_abs("/dev/partitions/PART_NAME")` are not mutually exclusive, for efficiency reason we need to keep using raw partitions on ESP32 (that are simple to mmap), so we need to support also  `code:load_abs("/dev/partitions/PART_NAME")`.

UncleGrumpy, 2023-01-22:
I was not thinking about the fact that it would have to be mmapped; but of course is does, or the file would be loaded into ram and consume far too much memory.

UncleGrumpy, 2023-01-23:
As far as a path name for the partitions maybe just:

`code:load_abs("/dev/mmap/PART_NAME").`

That seems to be enough to get the point across without being unnecessarily long… especially since that is ultimately how they are used in this case… mapped memory, more so than a traditional partition.

bettio, 2023-06-05:
#469 has been merged.

closed 2023-06-05 by bettio