atomvm/AtomVM #471: Provide better getting started guide.

dkuku, 2023-01-30:
Hi. Great project, but I had a hard time to compile an elixir file written by myself. I'm running it on linux.
Also compiling the example from command line not using the Makefile was very tricky for me, I could not find documentation for that and the examples are already compiled.
I ended up with editing the example and but I had to change `Console` to `:console` because after inspecting the generated file I had `Elixir.HelloWorld` and `console` atoms inside. Also I had to manually provide all dependant files to packbeam
```
elixirc examples/elixir/HelloWorld.ex
./build/tools/packbeam/PackBEAM hw.avm Elixir.HelloWorld.beam build/libs/eavmlib/src/beams/console.beam build/libs/eavmlib/src/beams/port.beam
./build/src/AtomVM hw.avm
```
This printed the result just like the compiled example but I imagine I wen't a bit over the board and there is an easier way.

dkuku, 2023-01-30:
ok, I found the examples repo https://github.com/atomvm/atomvm_examples/blob/master/elixir/README.md

closed 2023-01-30 by dkuku

UncleGrumpy, 2023-01-30:
It looks like you missed the missed the link to the AtomVM website in the Getting Started section of the README:
https://AtomVM.net

You will find extensive documentation there, including a very thorough Getting Started Guide…
https://doc.atomvm.net/0.6.0-dev/getting-started-guide.html

Note: the above link is for the “master” branch… a version for the 0.5 release can also be found on the [AtomVM website](https://atomvm.net).