atomvm/AtomVM #480: Building AtomVM for esp32c3 target

Emru1, 2023-02-12:
Hello, I'm trying to build master AtomVM for esp32c3.
I'm using espressif's idf docker image, v4.4.4.

So, I'm running `idf.py set-target esp32c3` in `src/platform/esp32` and then:

- If I run `make` then I receive following error:
```
App "atomvvm-esp32" version: 1
CC build/bootloader/esp_hw_support/compare_set.o
riscv32-esp-elf-gcc: error: unrecognized command line option '-mlongcalls'
make[2]: *** [/opt/esp/idf/make/component_wrapper.mk:298: compare_set.o] Error 1
make[1]: *** [/opt/esp/idf/make/project.mk:649: component-esp_hw_support-build] Error 2
make: *** [/opt/esp/idf/components/bootloader/Makefile.projbuild:47: /project/src/platforms/esp32/build/bootloader/bootloader.bin] Error 2
```
- Running `idf.py build` goes on for a while, near the end I receive following error:
```
/project/src/platforms/esp32/components/avm_sys/platform_nifs.c: In function 'nif_esp_sleep_enable_ext0_wakeup':
/project/src/platforms/esp32/components/avm_sys/platform_nifs.c:331:21: error: implicit declaration of function 'esp_sleep_enable_ext0_wakeup'; did you mean 'nif_esp_sleep_enable_ext0_wakeup'? [-Werror=implicit-function-declaration]
     esp_err_t err = esp_sleep_enable_ext0_wakeup(pin, level);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                     nif_esp_sleep_enable_ext0_wakeup
/project/src/platforms/esp32/components/avm_sys/platform_nifs.c: In function 'nif_esp_sleep_enable_ext1_wakeup':
/project/src/platforms/esp32/components/avm_sys/platform_nifs.c:349:21: error: implicit declaration of function 'esp_sleep_enable_ext1_wakeup'; did you mean 'nif_esp_sleep_enable_ext1_wakeup'? [-Werror=implicit-function-declaration]
     esp_err_t err = esp_sleep_enable_ext1_wakeup(mask, mode);
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                     nif_esp_sleep_enable_ext1_wakeup
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

I checked `src/platforms/esp32/components/avm_sys/platform_nifs.c`, it includes `esp_sleep.h` where those functions are defined.

So, my question is, is there any way to build AtomVM for esp32c3?

fadushin, 2023-02-12:
Hi @Emru1 

please see PR #450 .  That has a fix.

Emru1, 2023-02-12:
@fadushin 
thanks for your help :D

fadushin, 2023-02-12:
PR #450 is now merged `master`

closed 2023-02-12 by fadushin