atomvm/AtomVM #544: Can it be use with nifpp/erl_nif?

leondavi, 2023-05-04:
If you can please give a short example how to integrate cpp code in case nifpp/erl_nif aren't supported.

bettio, 2023-05-06:
For performance reasons AtomVM has a different API, that anyway it shouldn't be too different.
In order to implement a NIF you need to implement one or more functions like this one:
```c
static term nif_esp_nvs_erase_key(Context *ctx, int argc, term argv[])
{
}
```

you also need to define a struct such as:
```c
static const struct Nif esp_nvs_erase_key_nif = {
    .base.type = NIFFunctionType,
    .nif_ptr = nif_esp_nvs_erase_key
};
```

implement a NIF resolver function:
```c
272 const struct Nif *nvs_nif_get_nif(const char *nifname)
273 {
274     if (strcmp("esp:nvs_get_binary/2", nifname) == 0) {
275         TRACE("Resolved platform nif %s ...\n", nifname);
276         return &esp_nvs_get_binary_nif;
277     }
278     if (strcmp("esp:nvs_set_binary/3", nifname) == 0) {
279         TRACE("Resolved platform nif %s ...\n", nifname);
280         return &esp_nvs_set_binary_nif;
281     }
282     if (strcmp("esp:nvs_erase_key/2", nifname) == 0) {
283         TRACE("Resolved platform nif %s ...\n", nifname);
284         return &esp_nvs_erase_key_nif;
285     }
[...]
290     if (strcmp("esp:nvs_reformat/0", nifname) == 0) {
291         TRACE("Resolved platform nif %s ...\n", nifname);
292         return &esp_nvs_reformat_nif;
293     }
294     return NULL;
295 }

```

and to register your nif resolver function:

```c
REGISTER_NIF_COLLECTION(nvs, nvs_nif_init, nvs_nif_get_nif)
```

You can manipulate terms using functions defined in term.h.

leondavi, 2023-05-09:
Thanks for the answer.