atomvm/AtomVM #560: [wip] Run tests on feature/smp

bettio, 2023-05-13:
Following manual tests should be executed on real hardware using `feature/smp` and latest esp-idf version.
In case of failure, it must be compared with master branch and an issue should be reported.

Please, write down here in the comments if anyone is willing to take care of any of the following tests. 

ESP32 sockets:
- [ ] receive UDP packet in active mode
- [ ] receive TCP packet in active mode
- [ ] receive UDP packet in passive mode
- [ ] receive TCP packet in passive mode
- [ ] receive in binary mode
- [ ] receive in list mode
- [ ] send in binary mode
- [ ] send in list mode

bettio, 2023-05-14:
Trying to test `uart:read(UART)` but it doesn't work. See also #565.

UncleGrumpy, 2023-05-14:
STM32 does not have a digital read yet. 
Digital write is broken, but can be fixed with the merge of #447. I thought I opened an issue at the same time, but apparently for got so that is issue #570.

UncleGrumpy, 2023-05-20:
My mistake,  binary receive causes a Guru meditation error...

❌ TCP receive in binary mode

See Issue #582

UncleGrumpy, 2023-05-20:
✅ TCP send in binary mode.

#582 was a red herring...

UncleGrumpy, 2023-05-21:
The blink.avm from atomvm/AtomVM/examples/erlang/esp32/blink.erl, and both the AtomVM and atomvm_examples versions of the morse_sever suffer from the same bug (see Issue #584) but are functional with modifications.

UncleGrumpy, 2023-05-21:
✅ GPIO blink led
✅ GPIO interrupt
✅ GPIO digital read

as far as jitter goes there is some steady incremental time loss in the blink example, not sure it it warrants an issue or not. Here is an attached logic analyzer scan taken with 12Mhz sample rate:
[blink_100ms.sr.tar.gz](https://github.com/atomvm/AtomVM/files/11524277/blink_100ms.sr.tar.gz)

bettio, 2023-05-23:
> as far as jitter goes there is some steady incremental time loss in the blink example, not sure it it warrants an issue or not. Here is an attached logic analyzer scan taken with 12Mhz sample rate: [blink_100ms.sr.tar.gz](https://github.com/atomvm/AtomVM/files/11524277/blink_100ms.sr.tar.gz)

Can you attach the source for reference too?

UncleGrumpy, 2023-05-23:
Here is the Erlang source for the test I used to acquire the above logic traces.
[blink.erl.tar.gz](https://github.com/atomvm/AtomVM/files/11548034/blink.erl.tar.gz)

UncleGrumpy, 2023-05-24:
The rest of the network test results:

✅ receive TCP packet in passive mode

✅ receive UDP packet in active mode
✅ receive UDP packet in passive mode
✅ UDP receive in binary mode
✅ UDP receive in list mode
✅ UDP send in binary mode
✅ UDP send in list mode

✅ test morse_server.erl

UncleGrumpy, 2023-05-25:
> The blink.avm from atomvm/AtomVM/examples/erlang/esp32/blink.erl, and both the AtomVM and atomvm_examples versions of the morse_sever suffer from the same bug (see Issue #584) but are functional with modifications.

I have updated #584, it is not specific to feature/smp... the regression happened in `master` sometime in the last 12 months.

UncleGrumpy, 2023-05-25:
✅ STM32 GPIO blink led (+ check jitter).

Blink works on stm32, and **is not** affected by issue #584. The attached logic analyzer trace shows a very consistent time lag similar to the esp32. Is it approximately 3ms lost for each 100ms. Also attached is the Erlang source used for the test.
[stm32_blink_100ms.sr.tar.gz](https://github.com/atomvm/AtomVM/files/11561933/stm32_blink_100ms.sr.tar.gz)
[blinky_stm32.tar.gz](https://github.com/atomvm/AtomVM/files/11561934/blinky_stm32.tar.gz)

UncleGrumpy, 2023-05-28:
Both the udp_server_blink and tcp_server_blink will work if modified to use atoms rather than numeric values (0 or 1). See #584

tcp_client_esp32 sill suffers from #592.

UncleGrumpy, 2023-05-29:
✅ `tcp_client_esp32.erl` example. 
Fixed in master #595, so that will be fine after merging.

UncleGrumpy, 2023-05-30:
When #604 is merged the following examples work correctly:
✅ tcp_server_blink.erl
✅ udp_server_blink.erl

UncleGrumpy, 2023-05-30:
✅ UART write

closed 2023-05-31 by bettio