atomvm/AtomVM #582: TCP receive in binary mode on feature/smp on esp32 causes immediate crash of the VM

UncleGrumpy, 2023-05-20:
When first receiving an active TCP packet in `binary` mode causes an immediate Guru Meditation Error on esp32:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4200c2d1  PS      : 0x00060530  A0      : 0x82007a9d  A1      : 0x3fcf79c0
0x4200c2d1: do_send_socket_error at /home/winford/github/UncleGrumpy/AtomVM/src/platforms/esp32/components/avm_builtins/socket_driver.c:542

A2      : 0x3fce9e94  A3      : 0x3fcecefc  A4      : 0xff52d90a  A5      : 0x3fceb8c0
A6      : 0x3fcf79d0  A7      : 0x3fcf79d2  A8      : 0x3fcce260  A9      : 0x3fcad014
A10     : 0x3fceb8d0  A11     : 0x00000000  A12     : 0x3fcecdf0  A13     : 0x00000001
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0xff52d912  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff

Backtrace: 0x4200c2ce:0x3fcf79c0 0x42007a9a:0x3fcf7a10 0x42007c82:0x3fcf7a40 0x42069f0a:0x3fcf7a60 0x4206a0da:0x3fcf7a90 0x4206a1ba:0x3fcf7ab0 0x42013d21:0x3fcf7ad0 0x420081db:0x3fcf7b60 0x42001694:0x3fcf7b80 0x40388d7d:0x3fcf7ba0
0x4200c2ce: do_send_socket_error at /home/winford/github/UncleGrumpy/AtomVM/src/platforms/esp32/components/avm_builtins/socket_driver.c:542

0x42007a9a: sys_init_platform at /home/winford/github/UncleGrumpy/AtomVM/src/platforms/esp32/components/avm_sys/sys.c:211

0x42007c82: term_from_atom_index at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/term.h:573
 (inlined by) globalcontext_make_atom at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/globalcontext.h:300
 (inlined by) sys_get_info at /home/winford/github/UncleGrumpy/AtomVM/src/platforms/esp32/components/avm_sys/sys.c:338

0x42069f0a: smp_spinlock_unlock at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/smp.h:210
 (inlined by) scheduler_wait at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/scheduler.c:79

0x4206a0da: cp_to_mod_lbl_off at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/stacktrace.c:51

0x4206a1ba: stacktrace_create_raw at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/stacktrace.c:92

0x42013d21: scheduler_entry_point at /home/winford/github/UncleGrumpy/AtomVM/src/libAtomVM/opcodesswitch.h:2287 (discriminator 22)

0x420081db: smp_rwlock_rdlock at /home/winford/github/UncleGrumpy/AtomVM/src/platforms/esp32/components/avm_sys/smp.c:228

0x42001694: pthread_mutex_init at /home/winford/esp/esp-idf/components/pthread/pthread.c:531

0x40388d7d: systimer_hal_counter_value_advance at /home/winford/esp/esp-idf/components/hal/systimer_hal.c:114

ELF file SHA256: a5f29f3b45aea7ab

Initiating core dump!
I (14560) esp_core_dump_uart: Press Enter to print core dump to UART...
I (14568) esp_core_dump_uart: Print core dump to uart...
Core dump started (further output muted)
Received  24 kB...
Core dump finished!
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fcf7d24, name: 'pthread', GDB name: 'process 1070562596'

================== CURRENT THREAD REGISTERS ===================
exccause       0x1c (LoadProhibitedCause)
excvaddr       0xff52d912
epc1           0x40384c6b
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x4200c2d1          0x4200c2d1 <do_send_socket_error+169>
lbeg           0x40056f5c          1074098012
lend           0x40056f72          1074098034
lcount         0xffffffff          4294967295
[... 169 lines cut ...]
.coredump.tasks.data 0x3fcf5008 0x158 RW
.coredump.tasks.data 0x3fcf4d40 0x2c0 RW
.coredump.tasks.data 0x3fcf48a8 0x158 RW
.coredump.tasks.data 0x3fcf4610 0x290 RW
.coredump.tasks.data 0x3fcecb98 0x158 RW
.coredump.tasks.data 0x3fcec8a0 0x2f0 RW
.coredump.tasks.data 0x3fcf13a4 0x158 RW
.coredump.tasks.data 0x3fcf1100 0x290 RW
.coredump.tasks.data 0x3fcf1e0c 0x158 RW
.coredump.tasks.data 0x3fcf1b70 0x290 RW
.coredump.tasks.data 0x3fcae250 0x158 RW
.coredump.tasks.data 0x3fcadf70 0x2d0 RW
.coredump.tasks.data 0x3fcf4148 0x158 RW
.coredump.tasks.data 0x3fcf3ce0 0x460 RW
.coredump.tasks.data 0x3fcf2f6c 0x158 RW
.coredump.tasks.data 0x3fcf2cc0 0x2a0 RW
.coredump.tasks.data 0x3fcb086c 0x158 RW
.coredump.tasks.data 0x3fcb05a0 0x2c0 RW

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
Coredump checksum='50ba4af1'
I (16779) esp_core_dump_uart: Core dump has been written to uart.
CPU halted.

pguyot, 2023-05-20:
Can you reproduce the crash?
Could you please provide further details?

This is what I tried:
feature/smp branch (66dcbae31cff4a1d4bc33955ef44120da211353c)
The following change to AtomVM/atomvm_examples/erlang/tcp_server:

```diff
diff --git a/erlang/tcp_server/src/tcp_server.erl b/erlang/tcp_server/src/tcp_server.erl
index 7728203..2d3331f 100644
--- a/erlang/tcp_server/src/tcp_server.erl
+++ b/erlang/tcp_server/src/tcp_server.erl
@@ -25,7 +25,7 @@
 start() ->
     ok = maybe_start_network(atomvm:platform()),
     Port = maps:get(port, config:get()),
-    case gen_tcp:listen(Port, []) of
+    case gen_tcp:listen(Port, [binary, {active, true}]) of
         {ok, ListenSocket} ->
             io:format("Listening on ~p.~n", [local_address(ListenSocket)]),
             spawn(fun() -> accept(ListenSocket) end),
@@ -42,23 +42,29 @@ accept(ListenSocket) ->
                 local_address(Socket), peer_address(Socket)
             ]),
             spawn(fun() -> accept(ListenSocket) end),
-            echo();
+            echo(Socket);
         Error ->
             io:format("An error occurred accepting connection: ~p~n", [Error])
     end.
 
-echo() ->
+echo(Socket) ->
     io:format("Waiting to receive data...~n"),
     receive
-        {tcp_closed, _Socket} ->
+        {tcp_closed, Socket} ->
             io:format("Connection closed.~n"),
             ok;
-        {tcp, Socket, Packet} ->
-            io:format("Received packet ~p from ~p.  Echoing back...~n", [
-                Packet, peer_address(Socket)
+        {tcp_error, Socket, Error} ->
+            io:format("TCP error ~p.~n", [Error]),
+            ok;
+        {tcp, Socket, Packet} when is_binary(Packet) ->
+            io:format("Received packet of ~p bytes from ~p.  Echoing back...~n", [
+                byte_size(Packet), peer_address(Socket)
             ]),
             gen_tcp:send(Socket, Packet),
-            echo()
+            echo(Socket);
+        Other ->
+            io:format("Unexpected message ~p~n", [Other]),
+            echo(Socket)
     end.
 
 local_address(Socket) ->
```

The ESP32 monitor shows this:
```
I (2207) NETWORK: SYSTEM_EVENT_STA_CONNECTED received.
I (2237) wifi:AP's beacon interval = 122880 us, DTIM period = 2
W (2457) wifi:<ba-add>idx:0 (ifx:0, 8c:97:ea:e9:56:68), tid:0, ssn:1, winSize:64
I (5397) esp_netif_handlers: sta ip: 192.168.1.160, mask: 255.255.255.0, gw: 192.168.1.254
I (5397) NETWORK: SYSTEM_EVENT_STA_GOT_IP: 192.168.1.160
Acquired IP address: "192.168.1.160" Netmask: "255.255.255.0" Gateway: "192.168.1.254"
Listening on "0.0.0.0:44404".
Waiting to accept connection...
Accepted connection.  local: "192.168.1.160:44404" peer: "192.168.1.175:51589"
Waiting to receive data...
Waiting to accept connection...
Received packet of 12 bytes from "192.168.1.175:51589".  Echoing back...
Waiting to receive data...
Received packet of 54 bytes from "192.168.1.175:51589".  Echoing back...
Waiting to receive data...
Connection closed.
```

The client session was the following:
```
> nc 192.168.1.160 44404
Hello World
Hello World
Lookig at https://github.com/atomvm/AtomVM/issues/582
Lookig at https://github.com/atomvm/AtomVM/issues/582
> 
```

UncleGrumpy, 2023-05-20:
Indeed. Using the tcp_client and tcp_server from https://github.com/atomvm/atomvm_examples/tree/master/erlang/ and only adding `binary` to the options in both does work with the server running on esp32 and connecting with either esp32 or unix clients.  I am not sure why the test application I was using would crash when `binary` is added as an option, but it is clearly my own personal bug.

closed 2023-05-20 by UncleGrumpy