atomvm/AtomVM #592: Several examples in  `examples/erlang/esp32` do not work correxctly with `feature/smp` branch

UncleGrumpy, 2023-05-25:
The following examples all fail in a very similar manner immediately after acquiring an IP address: 

- udp_server_blink.erl
- tcp_server_blink.erl
- tcp_client_esp32.erl

Crash logs for each are attached. These examples were compiled at the same time as the `generic_unix` port and libraries, so they do not include the full stack trace information. I will recompile them with `packbeam -i` and see if I can find where the problem is.

[tcp_client_esp32.crash.tar.gz](https://github.com/atomvm/AtomVM/files/11562434/tcp_client_esp32.crash.tar.gz)
[tcp_server_blink.crash.tar.gz](https://github.com/atomvm/AtomVM/files/11562435/tcp_server_blink.crash.tar.gz)
[udp_server_blink.crash.tar.gz](https://github.com/atomvm/AtomVM/files/11562437/udp_server_blink.crash.tar.gz)

pguyot, 2023-05-27:
The problem is the socket is closed on the other end before the packet is processed and `inet:peername/1` is called a second time. I observed the same behavior with Erlang/OTP 25.

So I'll fix the example.

```erlang
{ok, Socket} = gen_tcp:connect("www.example.com", 80, [{active, true}]),
ok = gen_tcp:send(Socket, "GET / HTTP/1.0\r\n\r\n"),
receive Response -> io:format("~p\n", [Response]) end,
receive CloseMessage -> io:format("Got another message: ~p\n", [CloseMessage]) after 10000 -> io:format("No message for 10 secs\n") end,
io:format("Asking for peername after the socket is probably closed on the other end: ~p\n", [inet:peername(Socket)]),
{error, closed} = gen_tcp:send(Socket, "Hello again\r\n\r\n"),
io:format("Asking for peername after the socket was definitely closed: ~p\n", [inet:peername(Socket)]).
```

UncleGrumpy, 2023-05-27:
I was thinking this might be the case, but I didn't have time to investigate it yet. Thank you for the fix!

UncleGrumpy, 2023-05-28:
I closed this too soon, the tcp_client_esp32 still crashes after receiving a respone.

UncleGrumpy, 2023-05-29:
Fixed in with the merge of #595.

commit 032fdbb5 (2023-05-27): Fix a bug in ESP32 TCP client example (fix #592)

The example crashed on feature/smp branch (this is #592) because it called
`inet:peername/1` on a closed socket and received `{error, einval}` and did
not expect this return value. Yet this is the correct behavior.

Fix the example by calling `inet:peername/1` only once.

Also change the format strings so the packet will be printed properly once
bug #594 is fixed.
commit a3799b53 (2023-05-27): Merge pull request #592 from pguyot/w21/fix-tcp_client_esp32-example

Fix a bug in ESP32 TCP client example (fix #592)

The example crashed on feature/smp branch (this is #592) because it called
`inet:peername/1` on a closed socket and received `{error, einval}` and did not
expect this return value. Yet this is the correct behavior.

Fix the example by calling `inet:peername/1` only once.

Also change the format strings so the packet will be printed properly once bug

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual [... cut]
closed 2023-05-29 by UncleGrumpy