atomvm/AtomVM #633: Port ESP32 network driver to new netif API

bettio, 2023-06-11:
ESP32 network driver cannot be compiled with esp-idf 5 since deprecated network API has been removed (See also: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/networking.html#other-tcpip-adapter-api-replacement).

Timtonix, 2023-06-14:
Hey, I tried to build AtomVM with esp-idf.
Your project is awesome !
I've some errors during the build.
`HINT: All the Partition APIs have been moved to the new component 'esp_partition' - please, update your project dependencies. See Storage migration guide 5.x for more details` is that related to the issue ?

fadushin, 2023-06-14:
Support for IDF 5.0 is in the works (PR #626).  You can use IDF 4.4 or even just download a pre-built image from the lashes ESP32-mkimage builds off master.  E.g., https://github.com/atomvm/AtomVM/actions/runs/5253917200

Timtonix, 2023-06-15:
Ok thanks ! I'll do that thanks

commit 7ce49175 (2023-06-20): Merge pull request #639 from UncleGrumpy/esp_netif-update

Update esp32 network driver to use esp_netif

Updates the network interface from the depreciated tcpip_adapter to the
esp_netif component.

closes #633

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #639 (2023-06-20): Update esp32 network driver to use esp_netif
Updates the network interface from the depreciated tcpip_adapter to the esp_netif component.

closes #633

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2023-06-20 by bettio