atomvm/AtomVM #651: Improving SPI naming and consistency

bettio, 2023-06-28:
bus_config and device_config are defined as below:
```erl
-type spi_peripheral() :: hspi | vspi.
-type bus_config() :: [
    {miso_io_num, non_neg_integer()}
    | {mosi_io_num, non_neg_integer()}
    | {sclk_io_num, non_neg_integer()}
    | {spi_peripheral, spi_peripheral()}
].
-type device_config() :: [
    {clock_speed_hz, non_neg_integer()}
    | {mode, 0..3}
    | {spi_cs_io_num, non_neg_integer()}
    | {address_len_bits, 0..64}
    | {command_len_bits, 0..16}
```

The following might be renamed, since the sound redundant:
* `spi_peripheral` -> `peripheral`
* `spi_cs_io_num` -> `cs_io_num`

Also `hspi` and `vspi` sound like very specific to ESP32, but we should try to keep it more generic, since we should support also STM32 and RPI2040.

bettio, 2023-06-29:
Let's add an additional proposal:

```erl
-type bus_config() :: [
    {miso, non_neg_integer()}
    | {mosi, non_neg_integer()}
    | {sclk, non_neg_integer()}
    | {peripheral, spi_peripheral()}
].
```

I think that maybe shorter names are clearer and more readable.

UncleGrumpy, 2023-06-29:
> The following might be renamed, since the sound redundant:
> 
>     * `spi_peripheral` -> `peripheral`
> 
>     * `spi_cs_io_num` -> `cs_io_num`

`peripheral` is a bit misleading, this actually refers to a spi bus. I beleive `spi_bus`, or `bus_num` would be a more accurate name here.
`cs_pin` would be an even shorter, but clearer description of this setting.

> Also `hspi` and `vspi` sound like very specific to ESP32, but we should try to keep it more generic, since we should support also STM32 and RPI2040.

Newer versions of esp-idf refer to these as `spi0` and `spi1`, other boards may have more or less spi buses available, but using a number identifier would be the most platform agnostic way of labeling the buses.

commit 2cc65f70 (2023-12-16): Merge pull request #696 from bettio/spi-api-renames

SPI: API renames

Make names shorter and clearer, with no additional clutter.

Closes #651

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
PR #696 (2023-12-16): SPI: API renames
Make names shorter and clearer, with no additional clutter.

Closes #651 

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
closed 2023-12-16 by bettio