atomvm/AtomVM #736: Return `{ok, result}` tuple from NVS function

bettio, 2023-08-06:
Functions such as: `esp:nvs_get_binary/2` return either a binary or `undefined` atom.
The API might be improved to return instead `{ok, result} | error` rather than just the result.

In order to avoid compatibility issues, as alternative to a  return type change, `esp:nvs_fetch_binary/2` function might be implemented and the previous might be just deprecated, while the new might return the new result type.

UncleGrumpy, 2023-08-06:
Personally I would prefer we make a breaking change here and retain the current `nvs_get_binary` and `nvs_set_binary` function names. If we are going to replace them with the changed functions maybe both should be renamed, so that their names imply the same semantic relationship, like `nvs_write_binary` and `nvs_read_binary`.

bettio, 2023-08-06:
I don't mind keeping the function name as-is.
Anyway as an additional input, there is an interesting convention in Elixir:
- fetch() returns either `{ok, result}` or error
- get() returns either a valid result or a nil atom (that is same as undefined)

bettio, 2023-08-06:
One more question:
Right now we are doing:
```
 70             fprintf(stderr, "Unable to open NVS. namespace '%s' err=%i\n", namespace, err);
 71             RAISE_ERROR(term_from_int(err));
```
Should we keep raising errors for unexpected internal errors? Which ones should be critical and raise error?
Therefore should we prefer returning `{error, error_reason}` rather just `error`?

UncleGrumpy, 2023-08-06:
I think {error, error_reason} is always preferable, it makes the error more obvious, and allows the user to handle different error conditions programmatically.

UncleGrumpy, 2023-08-06:
> I don't mind keeping the function name as-is. Anyway as an additional input, there is an interesting convention in Elixir:
> 
>     * fetch() returns either `{ok, result}` or error
> 
>     * get() returns either a valid result or a nil atom (that is same as undefined)

Actually that sounds like a very interesting alternative. An argument could be made that returning an error might be overly presumptive anyway, returning `undefined` might be more useful. But having the `fetch` option could give the best of both. I assume `fetch` would also return `{error, Reason}` in the case of an actual error and just `error` in the case of and undefined entry?

bettio, 2023-08-11:
Fixed with #741

closed 2023-08-11 by bettio