atomvm/AtomVM #751: Build fails due to dialyzer error

bettio, 2023-08-11:
The following error is reported when building AtomVM:

```
[ 86%] Generating atomvmlib.plt
  Compiling some key modules to native code... done in 0m0.14s
  Creating PLT /path/to/AtomVM/build/libs/atomvmlib.plt ...
dialyzer: Analysis failed with error:
Could not scan the following file(s):
  Could not get Core Erlang code for: /path/to/AtomVM/build/libs/eavmlib/src/beams/logger.beam

Last messages in the log cache:
  Reading files and computing callgraph...
```

fadushin, 2023-08-12:
I think there is a (perhaps?) related issue, in that the ESP32 builds are broken because dialyzer is not on these machines/VMs/containers.

E.g.,
```
FAILED: libs/atomvmlib.plt /__w/AtomVM/AtomVM/src/platforms/esp32/test/build/esp-idf/main/test_erl_sources/HostAtomVM-prefix/src/HostAtomVM-build/libs/atomvmlib.plt 
ninja failed with exit code 1
cd /__w/AtomVM/AtomVM/src/platforms/esp32/test/build/esp-idf/main/test_erl_sources/HostAtomVM-prefix/src/HostAtomVM-build/libs && dialyzer --build_plt --output_plt /__w/AtomVM/AtomVM/src/platforms/esp32/test/build/esp-idf/main/test_erl_sources/HostAtomVM-prefix/src/HostAtomVM-build/libs/atomvmlib.plt -r /__w/AtomVM/AtomVM/src/platforms/esp32/test/build/esp-idf/main/test_erl_sources/HostAtomVM-prefix/src/HostAtomVM-build/libs/estdlib/src/beams -r /__w/AtomVM/AtomVM/src/platforms/esp32/test/build/esp-idf/main/test_erl_sources/HostAtomVM-prefix/src/HostAtomVM-build/libs/eavmlib/src/beams -r /__w/AtomVM/AtomVM/src/platforms/esp32/test/build/esp-idf/main/test_erl_sources/HostAtomVM-prefix/src/HostAtomVM-build/libs/alisp/src/beams
/bin/sh: 1: dialyzer: not found
```

from https://github.com/atomvm/AtomVM/actions/runs/5834358927/job/15823711962

bettio, 2023-08-12:
There was something dirty in my build directory, the problem is not there anymore. Closing this issue for now.

closed 2023-08-12 by bettio