atomvm/AtomVM #788: How was this put together?

mkschreder, 2023-08-25:
Would you mind sharing how you have put together this VM? I'm particularly interested to know whether this is an implementation from scratch or if it uses existing code from beam vm or somewhere else as base for its development.

fadushin, 2023-08-26:
It’s a ground-up implementation.  We run tests of Erlang/OTP APIs in order to ensure compatibility with OTP.  In some cases the OTP sources are consulted for comparison.

mkschreder, 2023-08-28:
@fadushin thanks for quick reply.

What do you think would be necessary to port the official otp to Zephyr RTOS?

fadushin, 2023-08-28:
I wouldn’t have good insight into that.  I think Peer Stritzinger would be a good source, as he has had to manage OTP on more esoteric platforms for his GRiSP board (though not RTOS, to my knowledge).  @pguyot may have some insights about POSIX compliance on RTOS-based platforms, such as ESP32.

pguyot, 2023-08-29:
I believe the first limiting factor would be RAM. GRISP has 64MB external SDRAM. It seems that ZephyrOS runs on many boards (including esp32) with too few RAM to run BEAM and for which AtomVM is a good alternative.

adolfogc, 2023-09-07:
Hi, is support for Zephyr currently in the works? 

Would it make sense to create a Zephyr module for AtomVM (https://docs.zephyrproject.org/latest/develop/modules.html)?

I'm interested in helping add support for Zephyr.

UncleGrumpy, 2023-09-07:
> Hi, is support for Zephyr currently in the works? 
> 
> 
> 
> Would it make sense to create a Zephyr module for AtomVM (https://docs.zephyrproject.org/latest/develop/modules.html)?
> 
> 
> 
> I'm interested in helping add support for Zephyr.

Zephyr is a very large and complex extra layer of APIs on top of the platform frameworks, I did briefly look at it when I started work on reviving the stm32 port of AtomVM. 

What MCU are you trying to target? We try to use the most direct and simplest toolchain and libraries to build off of, to keep the complexity down and be as performant as possible.

In my understanding Zephyr is like using an extra layer of api calls that then use the target systems Hal layer, that then eventually make their way down to low level api calls that get executed by the MCU. The other drawback to zephyr is its size and complexity to set up, especially for our CI workflows.

adolfogc, 2023-09-07:
Hi, thank you for your reply. I'm primarily interested in the ESP32-S3 SoC at the moment.

UncleGrumpy, 2023-09-07:
We already support all of the esp32 chips. For those I think it is safe to say that we will continue to use ESP-IDF for development, as any other option is just an extra layer of complexity on top of that.

UncleGrumpy, 2023-09-07:
> Hi, is support for Zephyr currently in the works? 
> 
> 
> 
> Would it make sense to create a Zephyr module for AtomVM (https://docs.zephyrproject.org/latest/develop/modules.html)?
> 
> 
> 
> I'm interested in helping add support for Zephyr.

I looked at your link, and I may have misunderstood your intentions. You were looking for a way to include AtomVM into an existing Zephyr project. I originally thought you were looking to make a port for an unsupported board using Zephyr as the base platform.
It looks to me like AtomVM would fall more into the category of a west project, rather than a zephyr module, but I have not personally used Zephyr. If you are familiar with Zephyr and would like to add support for AtomVM, I am sure we'd be happy for a PR.

adolfogc, 2023-09-08:
Hi, yes, I'm evaluating how to integrate AtomVM into a Zephyr project. In relation to this, I thought that the Zephyr port could be structured like a module.

> I originally thought you were looking to make a port for an unsupported board using Zephyr as the base platform.

One of the strengths of Zephyr is their wide hardware support. And by using device tree and its device driver model, adding new boards could require less effort. In that regard, projects developed for AtomVM could become more portable.

bettio, 2023-09-08:
Let's move the Zephyr releated discussion here: https://github.com/atomvm/AtomVM/issues/802

closed 2023-09-15 by bettio