atomvm/AtomVM #809: Support for ESP-S3 chip?

brucekissinger, 2023-09-15:
I am very new to AtomVM and wondered if it supports the ESP-S3 chipset?  I saw a comment in another issue that this chip was supported, but when I tried to flash the 0.5.0 AtomVM image using the command:

**esptool.py --chip esp32-s23 AtomVM-v0.5.0-ESP32.img** (note that some extra parameters are omitted), I get the response:

A fatal error occurred:  AtomVM-0.5.0-ESP32.img is **not an ESP32-S3 image**.  Should I try using the most recent AtomVM release?

Any advice for a newbie would be greatly appreciated

brucekissinger, 2023-09-15:
I'm sorry to post an issue.  I just read the release notes for the 0.6.0 image and noticed that the ESP-S3 is supported.  I downloaded the image and it flashed perfectly to the device

closed 2023-09-15 by brucekissinger

arpunk, 2023-09-15:
Hi @brucekissinger, you need to compile the image for esp32-c3 target:

```
cd src/platforms/esp32
idf.py fullclean
idf.py set-target esp32c3
idf.py build
./build/mkimage && ./build/flashimage.sh
```

I'm speaking from memory so instructions might differ.

UncleGrumpy, 2023-09-15:
> I'm sorry to post an issue.  I just read the release notes for the 0.6.0 image and noticed that the ESP-S3 is supported.  I downloaded the image and it flashed perfectly to the device

No problem, we didn't make the 0.6.0 tag very obvious... If you run into any other problems don't hesitate to open more issues. I believe we are working towards a more official release of 0.6 soon. Pointing out where the documentation needs clarification and finding any bugs is always a big help.