atomvm/AtomVM #810: Unable to make AtomVM

brucekissinger, 2023-09-15:
I am very new to AtomVM and must be doing something incorrectly.  I am trying to build AtomVM on a Linux machine and am following the _Getting Started_ guide.  I can successfully use cmake to build the various makefiles in the different subdirectories, but when I run the **make** command, it works for a while and then generates this error:

[  6%] Linking C executable ../../AtomVM
lib/liblibAtomVMLinux-i686.a(platform_nifs.c.o): In function `nif_crypto_hash':
/home/bruce/Projects/AtomVM/src/platforms/generic_unix/lib/platform_nifs.c:133: **undefined reference to `EVP_MD_CTX_new'**
/home/bruce/Projects/AtomVM/src/platforms/generic_unix/lib/platform_nifs.c:141: **undefined reference to `EVP_MD_CTX_free'**
/home/bruce/Projects/AtomVM/src/platforms/generic_unix/lib/platform_nifs.c:155: **undefined reference to `EVP_MD_CTX_free'**
/home/bruce/Projects/AtomVM/src/platforms/generic_unix/lib/platform_nifs.c:159: **undefined reference to `EVP_MD_CTX_free'**
collect2: error: ld returned 1 exit status
src/platforms/generic_unix/CMakeFiles/AtomVM.dir/build.make:89: recipe for target 'src/AtomVM' failed

If anyone has any advice it would be greatly appreciated.

fadushin, 2023-09-15:
You need to point `cmake` to OpenSSL3 headers and libraries on your platform.

E.g.,

```
cmake -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl -DOPENSSL_LIBRARIES=/usr/local/opt/openssl/lib ...`
```

We will need to make this more clear in the build instructions.  I think we also need to conditionally compile the offending code you hit, so I will file a ticket on that, as well.

Just out of curiosity, why are you trying to build AtomVM?

brucekissinger, 2023-09-15:
Thanks for the suggestion.

I was under the impression that I needed to build AtomVM in order to get an executable version of the PackBEAM application.  Here is what I have done so far.

1. Downloaded the ESP32 release image of AtomVM and flashed it to the chip.
2. Started following the steps listed in the _AtomVM Development_ guide.  
3. Wrote a hello_world.erl Erlang program
4. Need to use the PackBEAM app to package the compiled Erlang program

I see a packbeam.c file in the AtomVM/tools/packbeam folder and assumed that I needed to compile it somehow.  

Am I understanding this correctly?

UncleGrumpy, 2023-09-15:
We have a stand-alone version of packbeam available here:

https://github.com/atomvm/atomvm_packbeam

It looks like this should be made more clear in the documentation...

For compiling and flashing your own app there is a rebar3 plugin that can create an AtomVM application template, and flash your compiled app to ether an ESP32 or an STM32, available here:

https://github.com/atomvm/atomvm_rebar3_plugin

Hope this helps. Please follow up if you still run into problems.

UncleGrumpy, 2023-09-15:
I forgot to mention that the rebar3 plugin has packbeam included, so if you are using that the standalone packbeam application is not necessary.

fadushin, 2023-09-15:
Just to be clear, the `atomvm_rebar3_plugin` and the `atomvm_packbeam` code is all pure Erlang, so there is no need to build any C code out of the AtomVM repo.  The PackBEAM C program is used internally by AtomVM for packing the core libraries.

bettio, 2023-09-15:
@brucekissinger did @fadushin suggestion worked for you?
Also, can you tell us which Linux distribution are you using? Which version?
I think there is something we should improve on our side for sure.

brucekissinger, 2023-09-17:
I am using Ubuntu 18.04.06.

It sort of worked.  I was able to download the rebar3 plugin and build everything correctly.  I followed the instructions and created an AVM file named mylib.avm.  I was able to flash the AVM file but when I connect to the port, I get lots of messages that look like this:

Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048839
invalid header: 0xffffffff
....

fadushin, 2023-09-18:
Have you flashed the AtomVM VM image?  This sounds like an issue with not having an IDF application (in our case, the VM) flashed to the device.  Make sure to follow the instructions at https://www.atomvm.net/doc/master/getting-started-guide.html#deploying-the-atomvm-virtual-machine

Short of that, please feel free to contact us at any of the links on https://www.atomvm.net/contact/  We are most active on Telegram

fadushin, 2023-10-05:
@brucekissinger Can you give us an update on this ticket?  Thanks!

bettio, 2024-03-05:
I think the issue here is not relevant anymore (we even don't support OpenSSL anymore). Closing it.

closed 2024-03-05 by bettio