atomvm/AtomVM #817: Undecoded opcode: 172

mihailacusteanu, 2023-09-19:
I have an erlang app made using the instructions from here: https://www.atomvm.net/doc/master/getting-started-guide.html#getting-started-on-the-esp32-platform 

I use this code to flash the AtomVM
```
esptool.py --chip esp32 --port /dev/cu.usbserial-10 --baud 115200 \
    --before default_reset --after hard_reset \
    write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect \
     0x1000 AtomVM-v0.5.0-ESP32.img
```
     
and this one to flash my code:
     `rebar3 esp32_flash --port /dev/cu.usbserial-10`
my code is:
```erlang
-module(new_erlg).
-export([start/0]).

start() ->
    GPIO = gpio:open(),
    gpio:set_direction(GPIO, 2, output),
    loop(GPIO, off).

loop(GPIO, off) ->
    gpio:set_level(GPIO, 2, 0),
    timer:sleep(1000),
    loop(GPIO, on);
loop(GPIO, on) ->
    gpio:set_level(GPIO, 2, 1),
    timer:sleep(1000),
    loop(GPIO, off).
```
and the code gets flahed:

```
Flash will be erased from 0x00210000 to 0x00210fff...
Writing at 0x00210000... (100 %)
Wrote 16384 bytes at 0x00210000 in 1.6 seconds (82.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

when I use minicom to see my logs I get this

```

    ###########################################################

       ###    ########  #######  ##     ## ##     ## ##     ##
      ## ##      ##    ##     ## ###   ### ##     ## ###   ###
     ##   ##     ##    ##     ## #### #### ##     ## #### ####
    ##     ##    ##    ##     ## ## ### ## ##     ## ## ### ##
    #########    ##    ##     ## ##     ##  ##   ##  ##     ##
    ##     ##    ##    ##     ## ##     ##   ## ##   ##     ##
    ##     ##    ##     #######  ##     ##    ###    ##     ##

    ###########################################################

I (312) AtomVM: Starting AtomVM revision 0.5.0
I (312) AtomVM: Loaded BEAM partition main.avm at address 0x210000 (size=104857)
I (412) AtomVM: Found startup beam new_erlg.beam
I (412) AtomVM: Loaded BEAM partition lib.avm at address 0x1d0000 (size=262144 )
Undecoded opcode: 172
abort() was called at PC 0x4014de9c on core 0

Backtrace: 0x4009205c:0x3ffbbba0 0x40092289:0x3ffbbbc0 0x4014de9c:0x3ffbbbe0 0x0

Rebooting...
```

I am not sure what this `opcode: 172` means. I am coming from elixir and I may do something wrong.
The thing is that with the same AtomVM flashed if I flash my elixir code with `mix atomvm.esp32.flash --port /dev/cu.usbserial-10` it works.

fadushin, 2023-09-19:
Version 0.5.0 is vey old.  Can you try the latest 0.6.0 alpha release?

mihailacusteanu, 2023-09-19:
with 0.6.0 is working!
thanks a lot @fadushin !

closed 2023-09-19 by mihailacusteanu