atomvm/AtomVM #819: Need some guidance on getting started with Raspberry Pi Pico W

qhwa, 2023-09-22:
Hi, first of all, thank you for the library. It brings me the joy of Elixir to microcontrollers.

I'm struggling to get started with AtomVM and need some guidance.

I've read the [document](https://www.atomvm.net/doc/master/getting-started-guide.html#installing-atomvm-library-to-raspberry-pico) and tried to follow it. However, I never succeeded in running an application on Pico W.

I want to deploy the example _pico_blink_, here are some wondering:

> You typically need three uf2 files ...

1. How many `uf2` files should I copy to Rpi Pico, three or one?
    * I tried combining three uf2 files into one combined one, using `uf2tool join`, but after copying and rebooting, the USB device didn't appear again, nor did the LED blink.
    * I tried coping three uf2 separately, but after the first `AtomVM.uf2` was copied and an automatic rebooting, the microcontroller didn't appear again as a USB device. Should I enter the flashing mode with bootsel button again after the first and second write?
3. Should the `uf2` files created with additional attention?
    * For example, for the application uf2 file,  I tried `./uf2tool create -o pick_blink.uf2 pico_blink.avm` with or without `-s 0x100A0000` parameter, but none of them worked.

I apologize because I am getting started with two things simultaneously, Pico and AtomVM, and I don't understand how Pico works regarding its flashing pattern.

Looking forward to getting some ideas from you. Thank you.

qhwa, 2023-09-22:
Here's the device info by [picotool](https://github.com/raspberrypi/picotool) after copying both `AtomVM.uf2` and `atomvmlib.uf2` to the device:

```sh
$sudo picotool info -a

Program Information
 name:          AtomVM
 description:   Atom VM
 features:      USB stdin / stdout
 binary start:  0x10000000
 binary end:    0x1006ee14  # <-- I noticed that this doesn't change before and after copying the second uf2 file

Fixed Pin Information
 none

Build Information
 sdk version:       1.5.1-develop
 pico_board:        pico
 boot2_name:        boot2_w25q080
 build date:        Sep  1 2023
 build attributes:  Release

Device Information
 flash size:   2048K
 ROM version:  3
```

pguyot, 2023-09-22:
Thank you for your interest in AtomVM and its port on Raspberry Pi Pico.

You are doing things right. You need indeed to either combine the three UF2 files with `uf2tool join` (which is not documented) or flash them by putting the Pico in bootsel mode three times. The documentation can be improved to make this more explicit.

Each UF2 file includes the address where the data should be flashed, and you definitely need to pass `-s 0x100A0000` to uf2tool for your applications, otherwise you will overwrite data somewhere else.

The problem you are having actually is a regression introduced in 09fc0a49f618ad52476bc155d2ff8fb76a00bf31.
Good news is [a fix](https://github.com/atomvm/AtomVM/compare/master...pguyot:AtomVM:w38/fix-pico-build-tests) is in preparation. I'll submit a PR once CI passes and we'll close this issue when it's merged. Alternatively, you can try [latest alpha release](https://github.com/atomvm/AtomVM/releases/tag/v0.6.0-alpha.0) tag which works.

qhwa, 2023-09-22:
Thank you for the guidance @pguyot . 

I tried v0.6.0-alpha.0 which you mentioned and it works! I'm so excited to see the output coming out from the device. BTW I'm using the example of _hello_pico_ .

Really appreciate your help, and amazing work! 🥳

qhwa, 2023-09-22:
Another thing is that _pico_blink_ example still doesn't work. I think that's because the pin number used in the code is `25` which is incorrect. I'm checking the SDK to see which pin is expected.

closed 2023-09-23 by bettio

pguyot, 2023-10-05:
> Another thing is that _pico_blink_ example still doesn't work. I think that's because the pin number used in the code is `25` which is incorrect for Pico W. I'm checking the SDK to see which pin is expected.
> 
> Update: Pin number is 32 for Pico W.

I'm confused about this 32. I read that to access the board LED we should go through cyw43, which I created API for in #844 with a new picow_blink sample code, before adding support for networking. The documentation on the Pico W seems sparse and I may have missed it. Could you please provide a pointer?

qhwa, 2023-10-06:
I can't find the source now. There's no such definition in pico SDK.

It works in Arduino:

```c
void setup() {
  pinMode(32, OUTPUT);
}

void loop() {
  digitalWrite(32, HIGH);
  delay(1000);

  digitalWrite(32, LOW);
  delay(1000);
}
```

... but not in Micropython:

```python
>>> p = machine.Pin(32, machine.Pin.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid pin
```

So I think it's not correct.

pguyot, 2023-10-06:
Thank you. I found the source:
https://github.com/earlephilhower/arduino-pico/blob/master/variants/rpipicow/picow_digital.cpp#L47

We probably don't want to do that and stick to Pico W's official documentation.

qhwa, 2023-10-06:
That makes sense. Thank you for your efforts in supporting Pico W!

commit 7fbf05cb (2023-09-23): Merge pull request #820 from pguyot/w38/fix-pico-build-tests

Fix pico build (fix #819)

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit d249d2cc (2023-09-23): Fix pico build (fix #819)