atomvm/AtomVM #822: Unexpected crash

bettio, 2023-09-23:
```
I (815) AtomVM: Starting AtomVM revision 0.6.0-alpha.1+git.ffed16dd
I (825) sys: Loaded BEAM partition boot.avm at address 0x1d0000 (size=262144 bytes)
I (845) network_driver: Initialized network interface
I (845) network_driver: Created default event loop
I (865) AtomVM: Found startup beam esp32init.beam
W (865) sys: AVM partition not found for lib.avm
I (865) AtomVM: Unable to mount lib.avm partition. Hopefully the AtomVM core libraries are included in your application.
I (885) AtomVM: Starting esp32init.beam...
---
AtomVM init.
I (895) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting application...
Starting network...
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x000000bb  PS      : 0x00060230  A0      : 0x800d8981  A1      : 0x3ffc4810  
A2      : 0x000000bb  A3      : 0x3ffc7fe0  A4      : 0x3ffc7af8  A5      : 0x0000001f  
A6      : 0x3ffc7af0  A7      : 0x00000003  A8      : 0x800e1a89  A9      : 0x3ffc6438  
A10     : 0x3ffc7ad8  A11     : 0x0000001f  A12     : 0x3ffca8da  A13     : 0x0000000f  
A14     : 0x3ffc47ac  A15     : 0x3ffc47a4  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x000000b8  LBEG    : 0x400e1b7c  LEND    : 0x400e1b80  LCOUNT  : 0x00000000  
0x400e1b7c: context_clean_registers at AtomVM/src/libAtomVM/context.h:240
 (inlined by) scheduler_entry_point at AtomVM/src/libAtomVM/opcodesswitch.h:2108

0x400e1b80: scheduler_entry_point at AtomVM/src/libAtomVM/opcodesswitch.h:2110 (discriminator 7)

Backtrace:0x000000b8:0x3ffc48100x400d897e:0x3ffc48c0 0x400d16a4:0x3ffc48e0 
0x400d897e: scheduler_thread_entry_point at AtomVM/src/platforms/esp32/components/avm_sys/smp.c:65

0x400d16a4: pthread_task_func at esp-v4.4.2/esp-idf/components/pthread/pthread.c:195 (discriminator 15)
```

pguyot, 2023-09-24:
Interesting. Is it easy to reproduce?

#795 fixes how AtomVM interprets live, but I'm not sure it would fix the root cause of this crash.

bettio, 2023-09-25:
> Is it easy to reproduce?

yes and no. As soon as you have a the right stuff on the flash it happens 100% of times, but it is hard to predict how to cause it.

I discovered that if I remove the always start network from NVS the problem disappears, but it is hard to tell if it is just a coincidence.

bettio, 2023-09-28:
Update, it looks like the crash happens while executing the following code:

```
 
```

bettio, 2023-09-28:
The previous might be a red herring, I flashed a fresh build using the same flashed .avm and I've got the following:

```
AtomVM init.
I (928) sys: Loaded BEAM partition main.avm at address 0x210000 (size=1048576 bytes)
Starting network...
Starting application...
Warning: function lists:tuple_size/3 cannot be resolved.
CRASH 
======
pid: <0.4.0>

Stacktrace:
undefined

cp: #CP<module: 14, label: 55, offset: 39>

x[0]: error
x[1]: undef
x[2]: error

Stack 
------

I (1088) spi_driver: SPI Bus initialized.
[{connected,#Fun<erl_eval.15.34359738376>},{got_ip,#Fun<erl_eval.31.34359738376>},{disconnected,#Fun<erl_eval.47.34359738376>},{I (1088) spi_driver: SPI device my_device added.
ssid,<<"123456789">>},{psk,<<"qwertyuiopas">>}]
#CP<module: 14, label: 28, offset: 62>
<0.5.0>
#Ref<0.0.0.2>
{state,[{sta,[{connected,#Fun<erl_eval.15.34359738376>},{got_ip,#Fun<erl_eval.31.34359738376>},{disconnected,#Fun<erl_eval.47.34359738376>},{ssid,<<"123456789">>},{psk,<<"qwertyuiopas">>}]}],undefined,undefined,undefined}
{<0.3.0>,#Ref<0.0.0.1>}
#CP<module: 4, label: 90, offset: 54>
[]
{state,network,network,{state,[{sta,[{connected,#Fun<erl_eval.15.34359738376>},{got_ip,#Fun<erl_eval.31.34359738376>},{disconnected,#Fun<erl_eval.47.34359738376>},{ssid,<<"123456789">>},{psk,<<"qwertyuiopas">>}]}],undefined,undefined,undefined}}
{state,[{sta,[{connected,#Fun<erl_eval.15.34359738376>},{got_ip,#Fun<erl_eval.31.34359738376>},{disconnected,#Fun<erl_eval.47.34359738376>},{ssid,<<"123456789">>},{psk,<<"qwertyuiopas">>}]}],undefined,undefined,undefined}
{<0.3.0>,#Ref<0.0.0.1>}
#CP<module: 4, label: 125, offset: 0>

Registers
----------
x[0]: error
x[1]: undef
x[2]: error
x[3]: [{connected,#Fun<erl_eval.15.34359738376>},{got_ip,#Fun<erl_eval.31.34359738376>},{disconnected,#Fun<erl_eval.47.34359738376>},{ssid,<<"123456789">>},{psk,<<"qwertyuiopas">>}]
x[4]: []
x[5]: psk
x[6]: []
x[7]: []
x[8]: []
x[9]: []
x[10]: []
x[11]: []
x[12]: []
x[13]: []
x[14]: []
x[15]: []

Mailbox
--------

Monitors
--------
<0.3.0> ---> <0.4.0>

**End Of Crash Report**
CRASH 
======
pid: <0.3.0>

Stacktrace:
undefined

cp: #CP<module: 14, label: 18, offset: 45>

x[0]: exit
x[1]: {undef,{gen_server,call,[<0.4.0>,start]}}
x[2]: exit

Stack 
------

#Ref<0.0.0.1>
5000
start
<0.4.0>
#CP<module: 14, label: 18, offset: 45>
{ok,<0.4.0>}
#CP<module: 14, label: 7, offset: 100>
[]
[]
[]
15000
#CP<module: 2, label: 22, offset: 48>
[]
[]
[]
[]
[]
#CP<module: 2, label: 105, offset: 0>

Registers
----------
x[0]: exit
x[1]: {undef,{gen_server,call,[<0.4.0>,start]}}
x[2]: exit
x[3]: process
x[4]: <0.4.0>
x[5]: {undef,[]}
x[6]: undef
x[7]: []
x[8]: []
x[9]: []
x[10]: []
x[11]: []
x[12]: []
x[13]: []
x[14]: []
x[15]: []

Mailbox
--------

Monitors
--------

**End Of Crash Report**
Registers: [{reg_op_mode,"0x80"},{reg_fr_msb,"0xD9"},{reg_fr_mid,"0x61"},{reg_fr_lsb,"0x9A"},{reg_pa_config,"0x8C"},{reg_pa_ramp,"0x09"},{reg_pa_ocp,"0x2B"},{reg_lna,"0x03"},{reg_fifo_addr_ptr,"0x00"},{reg_fifo_tx_base_addr,"0x00"},{reg_fifo_rx_base_addr,"0x00"},{reg_fifo_rx_current_addr,"0x00"},{reg_irq_flags_mask,"0x00"},{reg_irq_flags,"0x00"},{reg_rx_nb_bytes,"0x00"},{reg_header_cnt_value_msb,"0x00"},{reg_header_cnt_value_lsb,"0x00"},{reg_packet_cnt_value_msb,"0x00"},{reg_packet_cnt_value_lsb,"0x00"},{reg_modem_stat,"0x00"},{reg_pkt_snr_value,"0x00"},{reg_pkr_rssi_value,"0x00"},{reg_rssi_value,"0x00"},{reg_hop_channel,"0x00"},{reg_modem_config_1,"0x88"},{reg_modem_config_2,"0x94"},{reg_symb_timeout_lsb,"0x64"},{reg_preamble_msb,"0x00"},{reg_preamble_lsb,"0x10"},{reg_payload_length,"0x01"},{reg_max_payload_length,"0xFF"},{reg_hop_period,"0x00"},{reg_fifo_rx_byte_addr,"0x00"},{reg_modem_config_3,"0x04"},{reg_fei_msb,"0x00"},{reg_fei_mid,"0x00"},{reg_fei_lsb,"0x00"},{reg_rssi_wideband,"0x00"},{reg_detect_optimize,"0xC3"},{reg_invert_iq,"0x27"},{reg_detection_threshold,"0x0A"},{reg_sync_word,"0x2B"}]

Done
```

bettio, 2023-09-29:
- This is happening because a call to an invalid gcbif function pointer is happening:
```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x000000bb  PS      : 0x00060230  A0      : 0x800d8981  A1      : 0x3ffc4810  
```

gcbif structs cannot be corrupted since they are in flash, so a pointer to something else is used when building the import table (`module_build_imported_functions_table`).

- The following:
```
Starting application...
Warning: function lists:tuple_size/3 cannot be resolved.
```

Implies some kind of issue with `module_resolve_function0`.

- Both functions relies on `valueshashtable_get_value`.

According to further investigations, the same atom index is inserted twice, so a wrong atom string is returned for a given index.

By adding `abort()` when inserting the same atom index twice, the following backtrace is given:

```
Backtrace:0x40081eca:0x3ffc47900x4008b62d:0x3ffc47b0 0x40093232:0x3ffc47d0 0x40142962:0x3ffc4840 0x40114ff1:0x3ffc4860 0x40115025:0x3ffc4880 0x40117953:0x3ffc48a0 0x4013535a:0x3ffc48c0 0x400d9620:0x3ffc4930 0x4011519c:0x3ffc4960 0x40117f71:0x3ffc4980 0x400e40d4:0x3ffc4ab0 0x400d9bd6:0x3ffc4b50 0x400d16bc:0x3ffc4b70 0x4008eb51:0x3ffc4b90 
0x40081eca: panic_abort at esp-v4.4.2/esp-idf/components/esp_system/panic.c:402

0x4008b62d: esp_system_abort at esp-v4.4.2/esp-idf/components/esp_system/esp_system.c:128

0x40093232: abort at esp-v4.4.2/esp-idf/components/newlib/abort.c:46

0x40142962: valueshashtable_insert at AtomVM/src/libAtomVM/valueshashtable.c:77

0x40114ff1: globalcontext_insert_atom_maybe_copy at AtomVM/src/libAtomVM/globalcontext.c:430

0x40115025: globalcontext_insert_atom at AtomVM/src/libAtomVM/globalcontext.c:402

0x40117953: module_populate_atoms_table at AtomVM/src/libAtomVM/module.c:89

0x4013535a: module_new_from_iff_binary at AtomVM/src/libAtomVM/module.c:265

0x400d9620: sys_load_module at AtomVM/src/platforms/esp32/components/avm_sys/sys.c:400

0x4011519c: globalcontext_get_module at AtomVM/src/libAtomVM/globalcontext.c:516

0x40117f71: module_resolve_function0 at AtomVM/src/libAtomVM/module.c:423

0x400e40d4: module_resolve_function at AtomVM/src/libAtomVM/module.h:300
 (inlined by) scheduler_entry_point at AtomVM/src/libAtomVM/opcodesswitch.h:1832

0x400d9bd6: scheduler_thread_entry_point at AtomVM/src/platforms/esp32/components/avm_sys/smp.c:65

0x400d16bc: pthread_task_func at esp-v4.4.2/esp-idf/components/pthread/pthread.c:195 (discriminator 15)

0x4008eb51: vPortTaskWrapper at esp-v4.4.2/esp-idf/components/freertos/port/xtensa/port.c:131
``` 

However this shouldn't happen since #812.

PR #826 (2023-09-30): Fix race condition in atom table for real
Previous fix (see #812) didn't fix an important corner case. This was causing a number of odd looking behaviors (see #822).

Closes #822.

These changes are made under both the "Apache 2.0" and the "GNU Lesser General
Public License 2.1 or later" license terms (dual license).

SPDX-License-Identifier: Apache-2.0 OR LGPL-2.1-or-later
commit e9d9a0fc (2023-09-30): Fix race condition in atom table for real

Previous fix (see #812) didn't fix an important corner case.
This was causing a number of odd looking behaviors (see #822).

Closes #822.
closed 2023-09-30 by fadushin